betaflight/betaflight #3056: Betaflight 3.1.7 Kiss Esc Sensor Issue - Omnibus f4

UltiFix, 2017-05-09:
My kiss telemetry isnt working from my kiss escs to my omnibus f4.... Betaflight settings attached.

All four of the kiss escs telemetry wire is going to uart 3 rx. Double checked and I didnt accidently solder a ground or something. Using dshot600
[image]
[image]
[image]

mikeller, 2017-05-09:
Can you set `debug_mode = esc_sensor` and check what values are reported for 'debug' 0 to 3 on the 'Sensors' tab?

UltiFix, 2017-05-09:
Yep, here is a screenshot. Battery is plugged in but not armed.
[image]

mikeller, 2017-05-09:
This is the legend for the debug values:

https://github.com/betaflight/betaflight/blob/master/src/main/sensors/esc_sensor.c#L76-L79

Doesn't look good, the feedback is triggered (Debug 0), but there are a lot of timeouts (complete packet never received, Debug 1), and even more CRC errors (packet had an error, Debug 2). It looks like something is interfering with the serial transmission. Check your wiring again, and if necessary check every single ESC in isolation, keeping an eye on CRC errors.

UltiFix, 2017-05-09:
> main/sensors/esc_sensor.c#L76-L79

I have one wire going to rx uart3, then that wire and the 4 esc telemetry
wires all twisted together and soldered. Is that a problem?

basdelfos, 2017-05-09:
Twisting wires TLM wires and one wire to Rx uart3 should work, I have that also on my omnibus quad

UltiFix, 2017-05-09:
Ok, also wondering does the rx and tx pads need to be bridged? they are not right now

basdelfos, 2017-05-09:
Nope, they only have to bridged on the KISS FC.

mikeller, 2017-05-09:
@UltiFix: From what you are saying, it sounds like your wiring is correct. But the high number of CRC errors in your log suggests that the data is not received properly.

Just a quick thought: Do you have another UART that is unused, and can you try switching ESC telemetry over to use that other UART?

UltiFix, 2017-05-09:
You read my mind, I was planning in changing to uart6 I'll have to
sacrifice smart port telemetry but I have osd anyway

UltiFix, 2017-05-10:
This is switched to uart 6 rx. Still doesn't show any voltage in BF. Assuming I messed up somewhere here...
https://i.imgur.com/Ch7UnEB.png

UltiFix, 2017-05-10:
My wiring is perfect, just checked again. Reflashed BF and no diff. Soldering each one up seperatly now.

UltiFix, 2017-05-10:
esc 4:
https://i.imgur.com/UtnEqjM.png
esc 3:
https://i.imgur.com/CbjqM37.png
esc 2:
https://i.imgur.com/v9FMNpM.png
esc 1:
https://i.imgur.com/Fo7JuGc.png

this was just arming and doing random stuff. Although one of my packs got really low toward the end of esc 2 and I had to switch for esc 1

mikeller, 2017-05-10:
Looks like ESC 2 is causing the CRC errors. Is there anything different with the wiring for this ESC?

UltiFix, 2017-05-10:
Not that im aware of. Its all heatshrinked but I have red for pwm and black for ground and tlm and the one I think is tlm is not connected to ground so it has to be tlm. I can cut all the heatshrink back.. Its just so pretty haha. Want me to test esc 2 again?

UltiFix, 2017-05-10:
Would it be worth flashing the escs again?

mikeller, 2017-05-10:
Flashing ESC 2 again was going to be one of my next recommendations. The other ones look fine. Have you checked if you've got a short between PWM and TLM?

UltiFix, 2017-05-10:
went to 1.08f and nothing. No shorts. The sensor pic is of all 4 again:
https://i.imgur.com/sfFCz4k.png

UltiFix, 2017-05-10:
on latest firmware:
https://i.imgur.com/Pka6jvX.png

basdelfos, 2017-05-10:
Do you have a spare esc to replace the 2nd esc, to rule out that the esc is faulty?

UltiFix, 2017-05-10:
No

UltiFix, 2017-05-10:
I may just have to survive with the built in adc sensor.

mikeller, 2017-05-10:
If you got one that's on-board that's probably a better solution anyway. I found there to be quite a bit of tolerance on the ESC sensor readings, and in contrast to the onboard ADC sensor, they can't be calibrated at the moment.

UltiFix, 2017-05-10:
Unfortanetly I would like to see the current used. Could I pull the current
off the other 3 and just use those? Better than nothing.

mikeller, 2017-05-10:
Yes, if your flight controller / PDB does not have a current sensing shunt then ESC sensing will give you data that you wouldn't have otherwise. It would be easy to change the code to allow it to work with only a subset of the ESCs connected. It was chosen not to do this because, in the case of current / consumption measurement, getting a value reported that is 0 <= x <= actual current / consumption isn't of much value, and would cause people to discharge their batteries too low if they trusted on this value being an actual representation of consumption.

UltiFix, 2017-05-10:
Yea, I understand. Any way to take the three values combined, divide it by
three then take that and use it as the data for the missing esc?

mikeller, 2017-05-10:
Sure could be done with custom firmware for your situation. The ESC sensing code is in here: https://github.com/betaflight/betaflight/blob/master/src/main/sensors/esc_sensor.c

UltiFix, 2017-05-10:
I havent learned C++ yet. Will next semester though. Will have to wait until then.

mikeller, 2017-05-10:
Sounds like a good goal to have to motivate you to get stuck into some real world coding. :-)

UltiFix, 2017-05-10:
Yup! One reason why I enjoy college. You get to learn :D Im also doing it in a way were im doing it as part of highschool so its free for now.

Im having another issue with yaw, I opened up a new one here.
https://github.com/betaflight/betaflight/issues/3065
You don't need to help me with it, just wondering if its related to this telemetry problem and if I should mention it. Thanks.

UltiFix, 2017-05-11:
Closed - May try and make custom firmware later

closed 2017-05-11 by UltiFix