betaflight/betaflight #3057: adc on f7

nathantsoi, 2017-05-09:
anybody know if adcs are working in f7?

(free resource lines removed)

anyfcf7:
```
# resource list
Currently active IO resource assignments:
(reboot to update)
--------------------
A01: MOTOR 3
A02: MOTOR 2
A03: MOTOR 4
A04: MPU_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A11: USB
A12: USB
B02: BEEPER
B06: LED 2
B07: LED 1
B08: MOTOR 1
C04: MPU_EXTI
D03: SDCARD_DETECT
D12: I2C_SCL 4
D13: I2C_SDA 4
E11: SDCARD_CS
E12: SPI_SCK 4
E13: SPI_MISO 4
E14: SPI_MOSI 4

Currently active DMA:
--------------------
DMA2 Stream 1: SDCARD
```

omnibusf7:

```
# resource list
Currently active IO resource assignments:
(reboot to update)
--------------------
A04: SPI_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A15: MPU_CS
B00: MOTOR 1
B01: MOTOR 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
D15: BEEPER
E00: LED 1
E02: SPI_SCK 4
E03: SDCARD_DETECT
E04: SDCARD_CS
E05: SPI_MISO 4
E06: SPI_MOSI 4
E09: MOTOR 3
E11: MOTOR 4

Currently active DMA:
--------------------
DMA2 Stream 1: SDCARD
DMA2 Stream 7: SERIAL_TX 1
```

i would expect something like this (f3):

```
# resource list
IO
A00: PPM
A04: ADC_BATT
A05: ADC_CURR
A06: MOTOR 1
A07: MOTOR 2
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: MOTOR 3
A12: MOTOR 4
B02: ADC_RSSI
B03: LED 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: MPU_EXTI
C15: BEEPER

DMA:
DMA2 Channel 1: ADC
```

kc10kevin, 2017-05-09:
ADC showed up correctly in the resource list (DMA) on my test FuryF7 (STM32F722) boards.  I have not actually tested any of the ADC functionality yet though.

MJ666, 2017-05-09:
The ADC is working on our AlienFlightNG F7 brushless prototype boards.

sambas, 2017-05-09:
seems ok, you need to enable them in cli.
`set current_meter = ADC
set battery_meter = ADC
`
https://www.dropbox.com/s/4nxdgkvdbuyfzbp/Screenshot%202017-05-09%2017.34.32.png?raw=1
Only issue I just noticed on AnyFC-F7, RSSI value moves with VBAT.

digaus, 2017-05-19:
Seems to work on Omnibus F7 when set via CLI. Trying to set it with the pre 3.2 configurator breaks it and I need to reflash.

mikeller, 2017-05-19:
@digaus: While I am a bit surprised that this change should break the configuration to a point that requires a firmware reflash, the pre 3.2 configurator (i.e. the one that is distributed through the Chrome web store) is not supposed to work with 3.2 (in particular not with the battery configuration), so your problems aren't really a bug.

digaus, 2017-05-19:
Yeah I noticed that. I also do not know why it breaks the monitoring but after trying to change it with the configurator I was not able to get a voltage or current reading even when set via CLI afterwards. But this should not be done anyways so it ist not an issue. Just wanted to point out that ADCs are working on the F7

closed 2018-04-21 by DieHertz