betaflight/betaflight #3059: BF 3.1.7 blackbox incorrect amperage values

bigb2000, 2017-05-09:
Dear guys,
i'm observing weird amperage values in blackbox log. I "calibrated" the sensor using a clamp amperage meter and the displayed values in OSD and Smartport Telemetry (they match exactly) seem correct.
Screenshot of BB log and actual config dump are attached.

[bb amperage incorrect.zip](https://github.com/betaflight/betaflight/files/987443/bb.amperage.incorrect.zip)
[image]

mikeller, 2017-05-09:
This is a log viewer issue, the log viewer is using the wrong scaling for amperage values.

Duplicate of https://github.com/betaflight/blackbox-log-viewer/issues/52

closed 2017-05-09 by mikeller

bigb2000, 2017-05-10:
Thanks Michael, maybe the logs help identifying the problem. Same mentioned in betaflight/blackbox-log-viewer#52 i have no problem reading logs with my second quad running on BF 3.0.0.

mikeller, 2017-05-10:
@bigb2000: The reason this works correctly with 3.0.0, but not with 3.1.7 is that the unit of the value that amperage is stored has changed between 3.0.0 and 3.1.7, since the underlying implementation for current measurement has changed. Nothing unusual, and the solution will be to change how these values are displayed in the log viewer based on the firmware version that is set in the header of the log file.

ledvinap, 2017-05-15:
@mikeller : Even better, add blackbox header field that will specify current scaling ..