betaflight/betaflight #3065: Quad pitches and roll when I yaw - Blackbox attached

UltiFix, 2017-05-10:
You can see between frame 173180 and 177916 I only am yawing yet my quadcopter changes the pitch and roll drastically. Makes flying this quad miserable.

Build:
Omnibus F4
Kiss 24A
Have it setup were the motors are all spinning oposite direction and the props are switched for better handling when I nail stuff and to keep my camera clean
X4R-SB
X210
Taranis
BF 3.1.7

rnelias, 2017-05-10:
take a look: https://github.com/betaflight/betaflight/issues/2852#issuecomment-300455775

rnelias, 2017-05-10:
I'm having a similar issue with exactly the same setup (you had not mentioned but, in my case, I'm running dshot600 8k/8k). After hard throttle inputs, it seems the FC mixes some commands and the quad make unpredictable moves

UltiFix, 2017-05-10:
You talking about the fpv angle? Mine was set at 30 and I changed it back to zero and it didnt make a difference. Mine happens constantly, regardless of throttle.

UltiFix, 2017-05-10:
Should I go back to 3.1.6? Had kiss in this quad and the fc died (Rough life, learned soldering on it) and I wanted to try bf again. Bad choice, should have done kiss again. So this quad has not been on 3.1.6.

rnelias, 2017-05-10:
yep, I've set it back to 0 and the problem is still there... 

Curious that you're also having problem with motor #2. Yesterday I sent a BB log with the problem happening in motor #2 but a folk in rcgroups, after analyzing my data, thought it was normal. Well... I was flying the quad and I have others that are working fine, thus, I do feel something weird with my setups using Omnibus F4 and KISS

UltiFix, 2017-05-10:
Oh yeah, were both over their. I posted but nobody responds. So I post here lol.

Im going to go to 3.1.6 and try. Should have time a bit later tonight. This is exactly why i left betaflight in the first place, I already have kiss esc telemetry problems and now this.

rnelias, 2017-05-10:
yep, I'm going back to 3.1.6 too

My other quad running litlee bee (MS/4k) with SPRF3 does not present these problems... but both using Omnibus F4, KISS 24a dshot600/8k with BF3.1.7 are weird...

UltiFix, 2017-05-11:
It may be fixed with 3.1.6. To dark outside so could only hover in garage abit. To hard to tell without goggles

StarrrLiteNL, 2017-05-11:
Could this be related to #3016 ?

UltiFix, 2017-05-11:
Maybe, but 3.1.6 didn't fix it. DVr footage with synced bb log coming
today.

UltiFix, 2017-05-11:
https://youtu.be/tXJfJmxb5hc

UltiFix, 2017-05-11:
[ME3.zip](https://github.com/betaflight/betaflight/files/994464/ME3.zip)
log attached, 2/2 is what you want

UltiFix, 2017-05-11:
Nevermind, this is behaving correctly. Just used to flying an h quad with the camera mounted farther forward

fftunes, 2017-05-12:
@UltiFix Frame and camera mounting point shouldn't make a difference. Only difference that would make sense would be a different camera angle.

You should test this in LOS hover pirouettes, would make it much more obvious if there was a problem.

UltiFix, 2017-05-12:
I did, it was fine. I don't know, I may just be losing my mind. Finals just finished so im wiped haha. Ill have time to fly at least 4 packs tomorrow. Im sure its fine, and its me. Maybe its different rates or something.

closed 2017-05-12 by UltiFix