betaflight/betaflight #3070: SITL compiles with warnings.

AndersHoglund, 2017-05-11:
SITL compiles with warnings:

    $ export V=0
    $ make clean_SITL SITL
    Cleaning SITL
    Cleaning SITL succeeded.
    
    Building SITL
    ./src/main/drivers/stack_check.c: In function âstackTotalSizeâ:
    ./src/main/drivers/stack_check.c:93:12: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
         return (uint32_t)&_Min_Stack_Size;
                ^
    ./src/main/drivers/stack_check.c: In function âstackHighMemâ:
    ./src/main/drivers/stack_check.c:98:12: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
         return (uint32_t)&_estack;
                ^
    ./src/main/drivers/io.c:356:23: warning: ISO C forbids zero-size array âioDefUsedMaskâ [-Wpedantic]
     static const uint16_t ioDefUsedMask[DEFIO_PORT_USED_COUNT] = { DEFIO_PORT_USED_LIST };
                           ^
    ./src/main/drivers/io.c:356:42: warning: ISO C forbids empty initializer braces [-Wpedantic]
     static const uint16_t ioDefUsedMask[DEFIO_PORT_USED_COUNT] = { DEFIO_PORT_USED_LIST };
                                              ^
    ./src/main/drivers/io.c:357:22: warning: ISO C forbids zero-size array âioDefUsedOffsetâ [-Wpedantic]
     static const uint8_t ioDefUsedOffset[DEFIO_PORT_USED_COUNT] = { DEFIO_PORT_OFFSET_LIST };
                          ^
    ./src/main/drivers/io.c:357:43: warning: ISO C forbids empty initializer braces [-Wpedantic]
     static const uint8_t ioDefUsedOffset[DEFIO_PORT_USED_COUNT] = { DEFIO_PORT_OFFSET_LIST };
                                               ^
    ./src/main/drivers/io.c: In function âIOInitGlobalâ:
    ./src/main/drivers/io.c:365:34: warning: comparison is always false due to limited range of data type [-Wtype-limits]
         for (unsigned port = 0; port < ARRAYLEN(ioDefUsedMask); port++) {
                                      ^
    ./src/main/drivers/io.c:368:31: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
                     ioRec->gpio = (GPIO_TypeDef *)(GPIOA_BASE + (port << 10));   // ports are 0x400 apart
                                   ^
    ./src/main/sensors/barometer.c: In function âbaroDetectâ:
    ./src/main/sensors/barometer.c:67:27: warning: unused parameter âdevâ [-Wunused-parameter]
     bool baroDetect(baroDev_t *dev, baroSensor_e baroHardwareToUse)
                               ^
    ./src/main/fc/cli.c:449:13: warning: âcliRepeatâ defined but not used [-Wunused-function]
     static void cliRepeat(char ch, uint8_t len)
                 ^
    Linking SITL
    /tmp/cc2ZcqSm.ltrans27.ltrans.o: In function `dyad_listenEx':
    /home/andho/projects/cleanflight/andwho/betaflight/lib/main/dyad/dyad.c:965: warning: Using 'getaddrinfo' in statically linked applications requires at runtime the shared libraries from the glibc version used for linking
       text    data     bss     dec     hex filename
    1147990    8628   72456 1229074  12c112 ./obj/main/betaflight_SITL.elf
    14.69user 1.51system 0:31.22elapsed 51%CPU (0avgtext+0avgdata 45436maxresident)k
    24400inputs+105400outputs (0major+829011minor)pagefaults 0swaps
    Building SITL succeeded.

This masks warnings from other targets, making it hard to detect new warnings. All target builds MUST be clean.

ledvinap, 2017-05-11:
There was PR some time ago addressing most/all of the warnings.

Also `-Wpedantic` is IMO overkill for CF/BF, it triggers on gcc features that are harmless and useful, but do not follow exact wording of the standard.

ledvinap, 2017-05-11:
Oh ... where is the PR, it was against cs8425 branch and did not get merged

https://github.com/ledvinap/cleanflight/commit/04158ef21a1bef5dac10c7c1c297eafcead0844e

AndersHoglund, 2017-05-11:
OK, great. What PR is that? Can not find it.

There are still some 6 warnings when removing `-pedantic` . Not sure what the implications would be to remove it. But sure, it may be overkill to be pedantic. However, I think we should try keep things clean even with -pedantic, removing it would be absolutely the last resort.

ledvinap, 2017-05-11:
@AndersHoglund : The problem with `-pedantic` is more philosophical - warnings in this group are not related to code quality, good practises or dangerous constructions. The warnings only state that C standard does not allow that (usually something that is considered useful under some circumstances)

One example is zero size array - it is very useful in variable-length struct - you can define:
```
struct packet {
   int type;
   int len;
   char payload[];
};
```
`sizeof(packet)` is size of packet header. C standard does not allow this, so you must declare size of payload nonzero and then  subtract `sizeof(pkt.payload)`

AndersHoglund, 2017-05-11:
OK, might be a case. But if I want the size of the header, I would express that as `sizeof(header_t)` .  I would not ask for the size of the packet. Something this:

    typedef struct  
    {
        int type;
        int len;
    } header_t;

    
    struct packet 
    {
       header_t header;
       char payload[];
    };

int header_size = sizeof(header_t);

`

closed 2018-04-21 by DieHertz