betaflight/betaflight #3076: COM numbers

smanginpoochunk, 2017-05-12:
What's the deal with some fc's not getting COM numbers? I've searched reddit, over and over, and don't know where else to look. Can someone please point me in the right direction to figuring out why some fc's dont get com numbers even though I have the proper drivers installed with the proper .hex files? I'm getting more and more turned off to this hobby due to the lack of help here, and I'm not trying to be a dick.

Arakon, 2017-05-12:
That sounds entirely like an issue with your windows setup, not related to betaflight software.

smanginpoochunk, 2017-05-12:
Even though all of my other boards (and spares) are still connecting properly? So far I've had this issue with one Naze32, one Lux race v1, and most recently a racewhoop v1. With the Naze, I had three separate boards, two of which had COM numbers and one did not, even though the newest of the three was the one without the number. (I'm not meaning any offense in any way, the most I do online is the occasional Reddit comment and FB. This whole 'talking to strangers online' thing is brand-new to me.

Arakon, 2017-05-12:
Especially the naze simply uses a CP2102 USB to serial chip. That should get a com port completely regardless of whether the naze or software behind it is functional.. in fact, just hooking up that chip to USB would get you a com port if windows is assigning the proper driver. Try different USB cables and if possible, try a different PC.

smanginpoochunk, 2017-05-12:
I have, out of all of the micro USB cables I have, only the one will work with FC's, and I have double-checked it multiple times today to ensure it's still working properly. I have also tried a second PC, and ended up with the same results with all of my boards working except for the Racewhoop. I'm at a loss, I have no idea why it would be just one mode on one board not working correctly.

blckmn, 2017-05-14:
If an FC is not connecting try reinstalling the driver with it connected. Some FCs use USB to serial chips others use virtual comm prts. For the VCP FCs you can quickly test connectivity by putting it into DFU mode. Then if you run zadig you should be able to see the FC in STM boot loader mode.

Some FCs use different USB device codes (depending if F7 or F3 etc). So even though one board works, another may not (without installing the driver again).

blckmn, 2017-05-14:
I'm closing this as it's specific to a PC. RC Groups or Slack for assistance

closed 2017-05-14 by blckmn