betaflight/betaflight #3079: 3.1.7 motor issue on CC3D with PWM receiver

szepnorbee, 2017-05-13:
Hi! I have an issue with CC3D. I uploaded the 3.1.7 firmware succesfully, but not all correct. On QuadX configuration if i select PWM receiver (one wire per channel) 3 of the 4 motor stopped working. If i select PPM input then all fine (all 4 motor activated) but i can't have PPM receiver. Help me please! This is a bug?

My config:
Copter: QuadX
ESC mode: PWM (not OneShot etc.)
Receiver mode: PWM (one wire per channel)
Chanel map: 2T1AER34

Thank you!

fftunes, 2017-05-13:
Could be this was solved in actual code but nobody cared to release a patch...

https://github.com/betaflight/betaflight/issues/2780
https://github.com/betaflight/betaflight/issues/2871

Sheynar, 2017-05-14:
I think it's on PWM reciever code:  it works on 3.0.1 (as it had PWM timer clash protection), and works "bad" at 3.1.x (as it had no PWM timer clash protection). Other input methods (PPM, SBUS, etc) are have timer clash protection. I've bought PPM reciever for my quad and all works fine.
#2177

szepnorbee, 2017-05-14:
I tested the latest build ( betaflight_3.2.0-1271-968d2c2.zip	) with same result. Not working with PWM receiver. Unfortunately i can't use PPM or other then PWM with my HobbyKink HK-T6 remote controller. I rolling back to 3.0.1

gingerman, 2017-05-15:
szepnorbee I can confirm your initial issue - although my board was unfortunately a clone CC3D ( which I did not know when I bought my RTF quad ).

Sheynar, 2017-05-19:
@szepnorbee I think you can add PWM -> PPM(sbus) decoder (physycal module) between your receiver and FC controller input. It's not good problem solve, but it cheaper then new remote controller and reciever pack.

abonyi91, 2017-06-05:
I can confirm this issue too. I configured my CC3D to PWM receiver and 3 motors did not work.

dustinb, 2017-06-29:
See my note on #2780.  Make sure channels 1-4 are not going to failover mode.  Possibly because of a minimum pwm signal being hit.   I was seeing nothing on channels 1-4 in receiver tab for 3.1.7 and going back to 3.0.1 it would work fine.

closed 2018-04-21 by DieHertz