betaflight/betaflight #3080: SP racing F3 EVO Motor 4 not spin if remap servo 1 in DSHOT even assign pin5 instead of pin4

sakata2014, 2017-05-13:
Hi All,

Interesting, I have assigned motor pin4 to pin5 and start all motors with DSHOT normally.
However, motor 4 no spin again if servo 1 enabled.

# resource
resource BEEPER 1 C15
resource MOTOR 1 A00
resource MOTOR 2 A01
resource MOTOR 3 A02
resource MOTOR 4 A06
resource SERVO 1 B01
resource PPM 1 A15
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11

DERBYJ, 2017-08-07:
I can't get any motors to spin up after installing BF 3.2 on my Evo.  I did get BF3.2 working on my Motolab boards after remapping pins but no luck on the SPR Evo yet.

jflyper, 2017-08-08:
Can't happen...

closed 2018-04-21 by DieHertz