betaflight/betaflight #3082: Led Strip on Kiwi F4 V2 not working

x3f3x, 2017-05-13:
Hello, I'm not able to make LED strip work with the Kiwi F4 V2 and Betaflight, I've installed the firmware from your website and soldered the led signal into the led port on Kiwi F4 V2, the led strip has voltage but when i actívate the Led strip on betaflight nothing happen, all Led remain off, what do i have to do?

I do not know if I have to write the command or configure something in ports tab, to make led strip work?

Thank you for your time.

DeepnineRC, 2017-05-13:
use VTX.DTA pin for LED strip output

x3f3x, 2017-05-14:
Ok, solved, @DeepnineRC you were right.
 Can I ask you why they put a pin especific for the led strip and I have to use another pin? Is any reason for that?

Thank you for your time, you helped me a lot!!

DeepnineRC, 2017-05-14:
i don't know but see #2858

x3f3x, 2017-05-14:
Ok thank you so much!!

closed 2017-05-14 by x3f3x