betaflight/betaflight #3083: Small OSD improvements

halninekilo, 2017-05-13:
Hi,

some fellow pilots suggested a few very basic OSD improvements to me and I started implementing them...

- Add a pilot name element so the craft-name doesn't have to be misused for this function any more
- Switch ontime to flytime once armed - this element can do double duty saving screen estate _(...create a new 'on/flytime' element for this?)_
- Display a static "DISARMED" warning for added safety
- Enable flashing "LOW VOLTAGE" warning by default

[image]

I'd glady finalize and contribute this if there is any interest.

Cheers!

DanNixon, 2017-05-15:
Why is the use of the craft name in the OSD a misuse?

mikeller, 2017-05-15:
@DanNixon: I think @halninekilo means that the 'craft name' field is 'misused' by pilots to enter their name, with the intention of generating video footage with their name.

I don't know if having two distinct fields because of this is worth the additional clutter / space requirement. Maybe we should just change the field label in configurator to 'Name'?

DanNixon, 2017-05-15:
@mikeller Agree, "Name" would be a better label for it. I can only think the issue is the fact that "Craft Name" relates to the aircraft not the pilot.

Is the name actually used anywhere else?

lostcontrol, 2017-05-15:
@DanNixon I think it was first implemented to identify blackbox logs. That value shows up in blackbox logs so there it makes sense to be craft name and not pilot name.

mikeller, 2017-05-15:
The name is also used to generate name suggestions for blackbox logs and config backups. But those are suggestions only, so if they want to use the field to generate 'branded' video, **and** want to store files by name, they have to do some typing. :-)

lostcontrol, 2017-05-15:
@mikeller the craft name is present in the blackbox header.

`BLACKBOX_PRINT_HEADER_LINE("Craft name", "%s",                       systemConfig()->name);`

mikeller, 2017-05-15:
@lostcontrol: That's true. But then, if users want to have their name show up on the video **and** use the header information to identify log files, they can always use something like `<pilot name> / <craft name> as name.

lostcontrol, 2017-05-15:
@mikeller Not sure what you mean with "as name". I guess you mean something like "pilot name/craft name". Thinking out loud, what about adding support for a separator that will be handled in the OSD code to split the name as pilot and craft name?

Like if you have "Gugus/Alien5", the separator being '/', the OSD would print just "Gugus". Don't even think it would be required to have the craft name on the OSD. Not really nice tough and I think the craft name is currently limited to a pretty short length.

Would be likely nicer to have both craft name/pilot name (call sign) as two independent values.