betaflight/betaflight #3085: Betaflight OSD not showing VTX Channel

BigTomFPV, 2017-05-14:
As above I've turned on the VTX Channel and positioned it accordingly to show on the OSD but it isn't showing up.

Using DYS F4 board, running 3.1.6 with an immersionRC tramp

DeepnineRC, 2017-05-14:
#2254  will come soon
at the moment it works only with internal vtx modules

DanNixon, 2017-05-18:
@jflyper How far did you get with #2254? I can't remember what the issue keeping it from being merged was (othe rthan the conflicts, of course).

johnmoto1, 2017-08-20:
I have kiwi F4 v2 inbuilt osd and it does not show the vtx info

CC210, 2017-12-09:
Hglrc v5 pro not showin as well hadn't had time to check options like ports yet but doesn't work with just the vtx turned on in osd only dashes.

jflyper, 2017-12-09:
@CC210 I suspect it is a configuration issue.

closed 2017-12-09 by jflyper