betaflight/betaflight #3086: Unify Pro V1 3.1.7 bug?

Stavr0s91i, 2017-05-14:
I have a TBS Unify 5v ,V1 connected on my OmnibusF4 running 3.1.7 .Working fine.As soon as i enable Softserial (pwm 5) and i'm able to get the fc and the vtx comminicating but as soon as i power up the quad vtx boots up for a few seconds i see the betaflight logo on the pre-set frequency and then dissapears.I can change all the settings via taranis telemetry and i can see there is a communication between the fc and the vtx.But something seems to be screwing with the vtx(software wise) and i am unable to get any image from the vtx even if i scan all the frequencies.When i disable the softserial , vtx works fine so i know its software related.

lasharor, 2017-05-14:
https://github.com/betaflight/betaflight/issues/2846

Stavr0s91i, 2017-05-14:
Great! I've searched but seems i didnt search enough! Closed as it already exists thanks!

closed 2017-05-14 by Stavr0s91i