betaflight/betaflight #3087: Spek rx not reconized on KISS CC AIO beta 3.16 and 3.17

legacy13, 2017-05-14:
i have tried every combination of settings on 3.16 and 3.17 none of the combinations work. when i flah it back to kiss firmware spek sat works fine. Please fix this bug so i can try beta on this board PLEASE borris PLEASE :)

legacy13, 2017-05-14:
talked to borris and he said spektrum may not be enabled on the correct pin. does anyone know how i would go about correcting this? the standard kiss fc has a separate spektrum port but this board does not so maybe thats the issue.

mikeller, 2017-05-14:
From looking at the definition of the KISSFC, it seems to be expecting the Spektrum RX on UART2. Try connecting your RX to the UART2 RX pin (or if the pin labeling does not match the UART numbers, try all RX pins in turn to find the one that works).

legacy13, 2017-05-15:
@mikeller dude... how did i not think of this. you are the man. ive had 2 builds sitting for a month because of this. moved the signal from the dedicated serial in pad to rx2 and it works. pinout in beta must be wrong probably matching the standard kiss fc that has a seperate spektrum port 
Thank you

mikeller, 2017-05-15:
You're welcome. I think the difference in port numbering / labelling is caused by KISS using 1, 2, 3, whereas Betaflight uses the actual port numbers as used by the MCU, that are usually not consecutive.

closed 2018-04-21 by DieHertz