betaflight/betaflight #3088: SD Card causing loss of control: Omnibus F4 V3

MiddleMan5, 2017-05-14:
_Pertinent info:
Board: [Omnibus F4 V3 ](http://www.readytoflyquads.com/flip-32---f4---omnibus-v3)
Betaflight 3.1.6
2GB SD Card formatted FAT with 32KB allocation units
ESC comm: DSHOT 600
FRsky SBUS receiver_

Okay, so this is a strange one, I think I've been having these issues for awhile now, but never chalked it up to be the SD Card. It doesn't help that every time I had one of these lockups my craft would smash into something hard and break something, leading me away from the real issue. 

Troubleshooting:
- RSSI values stay green the entire flight, current draw and voltage levels all within nominal range.

Symptoms: 

- Craft will periodically boot up but not initialize ESCs (no startup beacon) 
- Craft will fly for a short amount of time (30 sec) then lock up (air-mode still engaged, motors still running) and fall out of the sky
- Black-box logs will show multiple empty files (can we please fix the time-stamps on these!!)

Replication:
This is the fricking tricky part, I have no real way to reproduce these consistently, nor is there functionality (that I know of) to diagnose these issues. 

- Place a full sdcard in omnibus
- plug battery in
- fly and hope

Fix:
- Cleared SD card flies fine

But to say the above is a fix isn't the whole truth, sure it fixes the symptoms of an overloaded card, but betaflight should never prioritize SD card communication over actual operation, there's a bug in the code here somewhere.

blckmn, 2017-05-14:
This is classic sdcard write issues causing erratic behaviour in the quad. I think we need to detect write issues (timeouts etc) and disable the sdcard logging.

MiddleMan5, 2017-05-14:
Maybe a feature switch, because personally I'd rather it start overwriting earlier logs and give me feedback on my latest flight. An OSD value might be handy to warn the user of an overflow.
If FREESPAC.E is used to allocate write memory, can't you check FREESPAC.E is not below some critical value, if it is check user options for halt/overwrite, choose lowest value log file (assuming it wasn't edited by the user) and add it's memory map to the FREESPAC.E file?

BlackJohnny, 2017-05-20:
I have found this issue because I was looking for a similar behavior. My quad uses Thyphoon 30A with DSHOT600, Omnibusf4 Pro, T-motor F40 2600KV and in my case the drone was flying ok for about 1 minute and then suddenly rolls back always on the same side (describing a parabola path). Because of this I started to change motors but I did not fix the issue, it did the same but on the other side. Eventually I have ended up changing all the motors but without fixing the problem. After all these I decided to try on DSHOT 300 and this was a somewhat good decision. The issue lessened (not happening every flight) however often you would hear some weird motor desync sounds and the drone would change for a second its fly attitude.

Now the weird part .... I was always logging the flight and always the log stopped just before the incident so I could not see any anomaly in the logs. I suspected that due to the motor the FC reboots or get frozen.

Investigating the backlog settings I have noticed that it was set to 100%. Thus, I have changed it to 25% (4khz) and this fixed the behavior entirely. Being a new drone I didn't have the will to go back on DSHOT600 and test. I was lucky to not brake a lot during these crashes and I wanted to enjoy it for a while ... eventually I will test it with DSHOT 600.

**And one important note: during my initial flights although I had the card in the drone, I was not using the logging because I didn't configure any switch to start it. So basically the drone did this even without logging.**

isilcala, 2017-07-19:
@BlackJohnny not configuring a switch to BlackBox means it will be 'always on'. the other note is the frequency should not exceed 2K or most of the sd cards would not be able to catch up. Typically set it to 1K is a good balance spot.

closed 2018-04-21 by DieHertz