betaflight/betaflight #3089: SD Card Timestamps

MiddleMan5, 2017-05-14:
I just wanted to post this issue as well, the lack of time stamps on the SD card logs are kind of annoying, any chance for a fix?

Or... maybe it's just me, in that case: Omnibus F4 V3, V2 Pro, and KiwiF4 all show this issue.

blckmn, 2017-05-14:
There is no real time clock on FCs. So there is no concept of time. The FC is reset each time you power it up. The best you can hope for is an incrementing file count, based on what is already on the sdcard.

blckmn, 2017-05-14:
Sorry - I'll close this - if you feel you want to discuss, then we can reopen.

MiddleMan5, 2017-05-14:
Erg. You're right, dumb

martinbudden, 2017-05-14:
@blckmn , actually you can do this. All you need is for the transmitter to send the time. It would certainly be possible to create a Taranis script to do this. See issue https://github.com/betaflight/betaflight/issues/2729

closed 2017-05-14 by martinbudden

blckmn, 2017-05-14:
Great idea. Though it needs a smart port connection, but would be doable - at least an initial power on time.

mikeller, 2017-05-14:
Another viable clock source would be a GPS (if connected).

martinbudden, 2017-05-15:
Indeed. Which is usually the case for iNav.

markandkymward, 2017-09-02:
Has there been any movement on this?  I'm actually interested in GPS time stamping because I plan to compare the on-board aircraft state data to some environmental data I'm collecting.  GPS time stamping would sync this nicely.

mikeller, 2017-09-03:
@markandkymward: I don't think anybody is working on this. But if you want to implement it, and submit a pull request for it, this will be highly appreciated.