betaflight/betaflight #3093: Unable to flash NucleoF7 from BFC

jrsmiley, 2017-05-15:
I have a Nucleo-F767ZI that I would like to flash with BF, however, I get " No configuration received within 10 seconds, communication failed" when attempting to connect to the /dev/tty.usbmodem1443 device.  The board is currently programmed with a demo "flashy".  Does BFC understand how to flash using the ST-Link that comes with the board or do I need to install something else before BFC will be able to flash BF?  I'm using the latest Chrome BFC 3.1.1.

Thanks!

DieHertz, 2017-05-15:
Your Nucleo board is not flashed to BF, hence `No configuration received` is a perfectly fine behavior.

The NucleoF7 target was designed for `Nucleo-F746ZG`, hence you'll have to perform some modifications in order to make it work on `Nucleo-F767IZ`.

To actually flash it, you should either use STLink USB or User USB and switch the board into DFU mode and flash with `No reboot sequence` checkbox selected.

jrsmiley, 2017-05-15:
Sorry to be a dense Newb, but I've looked everywhere I can think of to find out how to put the Nucleo into DFU mode.  I found a few articles, but nothing that comes right out and tells you how to do it.  I've even tried connecting the board to a Windows 10 PC and using the dfuse demo from STM.  It doesn't see a DFU target either.  I can use J-Link just fine, but not DFU.  I've also tried holding down the reset button while plugging the board into the J-Link usb.  No joy.  

Any help would be greatly appreciated.

DieHertz, 2017-05-15:
[image]

DieHertz, 2017-05-15:
The BOOT pin (BOOT0) should be marked by a silkscreen label on your Nucleo board.

closed 2017-10-23 by jflyper