betaflight/betaflight #3094: VTX Power/Pit control via switch

BigTomFPV, 2017-05-15:
As a lot of use tramps and unifys would it be possible to implement the smart audio to control vtx power by a switch. 

Say for example I use a 3 position switch on Taranis, all the way up being disarmed and vtx in pit mode, middle position vtx at set power and disarmed, all the way down arms quad. 

Just a thought :)

mikeller, 2017-05-16:
Duly noted. At the moment we are out of free ids for switches, so we'll first have to do something to extend the number of possible switch ids bejond 32, but then it's certainly an option.

jflyper, 2017-05-16:
Something similar is possible, but please do remember limitations posed by each different breed of vtxes. For example, SmartAudio doesn't go back into pit mode unless power cycled.

BigTomFPV, 2017-05-16:
That's fine sure once people start asking the question and it becomes public knowledge it won't be an issue

wolffman122, 2017-05-17:
mikeller  I was looking at this issue and was going to try working on it.  I am a software engineer with windows and Win CE.  Just wondering why there is a limit on free ids for switches.  Looking at rc_adjustments.h the first enum couldn't we just add another one there?  Sorry if I am missing something.  I know it might be a limitation to the chip that I don't know about yet?

mikeller, 2017-05-17:
Thanks @wolffman122 for offering up to do work on Betaflight, much appreciated. The limitation seems to be in how the flight modes are communicated over MSP: https://github.com/cleanflight/cleanflight/blob/master_v1.x/src/main/fc/msp_server_fc.c#L387

Recent discussion on the topic on this PR: https://github.com/betaflight/betaflight/pull/3074

wkochFPV, 2017-05-21:
I wanted to have this feature also. My LUA script takes care of this.

Requirements:
Same as for betaflight-tx-lua-scripts (OpenTX 2.2 compiled with LUA support, smartport telemetry connected to fc, unify pro connected to fc,, betaflight 3.1.x). My code is derived from these scripts.
Unlocked Unify Pro, if more than 25mW desired.

How to:
1.	Install betaflight-tx-lua-scripts as usual (my script only needs the file /SCRIPTS/BF/msp_sp.lua from the package located in /SCRIPTS/BF)
2.	Install my script VTXprog.lua in /SCRIPTS/MIXES
3.	In model menu of your taranis go to LUA-scripts page and enable VTXprog.lua and configure as follows:
4.	Enter your desired band: 1="A", 2="B", 3="E", 4="F", 5="R" (unify pro!)
5.	Enter your desired channel: 1,2,3…
6.	Select “global variable 1” (GV1, or another free global variable) as input for TxPower
7.	Now configure any switch to adjust GV1 to the desired power level: 0=pit mode, 1=25mW, 2=200mW … (use special functions menu, select ‘adjust GV1’ as action, enter desired value). Use e.g. 0 for your switch neutral position, 1 for switch up, 2 for switch down, if you would like to select 25mw or 200mw based on switch position.

Now you can:
Configure vtx in betaflight once to OPmodel “race” >> vtx starts in pit mode.
Power up your model and your taranis, put swich in neutral position, vtx is in pit mode.
Move you switch to change GV1 from 0 to 1 (or whatever you desire).
Script will automatically set band, channel, selected power and turn off pit mode -> vtx starts transmission
File “eng.wav” will be played as confirmation.
If you change your model battery, you can leave your taranis powered on, move switch back to neutral position (GV1->0) and then to the desired power level, the change will trigger the script to reprogram your vtx / start transmission again.

Without any warranty!!!! Hope this helps. Thank to the developers of betaflight-tx-lua-scripts for providing the necessary code to accomplish this task!

Script: 
[VTXprog.zip](https://github.com/betaflight/betaflight/files/1017219/VTXprog.zip)

mikeller, 2017-05-21:
@wkochFPV: Thank you very much for creating this script and sharing it with fellow flyers. We have got a repository that is dedicated to tx lua scripts: https://github.com/betaflight/betaflight-tx-lua-scripts

Would you mind opening a pull request to have your script added to the collection that is already in there, so that others can use it as input for their work?

BigTomFPV, 2017-05-29:
@wkochFPV Just attempted to set it up but when I put it in the mixes folder it isn't visible on the lua selection on telemetry page, even renaming it to less than 4 characters doesn't show. If I copy it into the telemetry folder it becomes visible but get an error when attempting to run it

wkochFPV, 2017-05-29:
@BigTomFPV Copy to MIXES folder. The script will NOT show up in the telemetry/display settings page. It can be accessed on a separate page ('lua scripts', page 12/14 in model settings, there is a list 'LUA1, LUA2, ...' select LUA1 and assign the script). Please make sure to use OpenTx 2.2 with LUA in compile options. Page 12/14 of model settings is only available on the taranis itself, not in the OpenTx companion software.

JamesCordell, 2017-06-29:
Hi

Sorry if someone has already suggested this idea before. Would it be a good idea to link the RSSI of the RX to the VTX power. The effect would be that the VTX steps up power as the RSSI reduces. This would result in automatic power selection.

wkochFPV, 2017-07-02:
@JamesCordell You could try my LUA script, if you own a taranis. The script does not have to be modified for your purpose. Instead of using a switch to define the desired power level  to "GV1" (global variable 1), simply define "special functions" in OpenTX to change "GV1" according to the measured RSSI. The script will automatically update your vtx when GV1 is modified. Please tell us, whether this works for you and what RSSI levels you found suitable for each vtx power level!

robert-hoffmann, 2017-08-16:
Sounds pretty cool !

Personally i like it starting in pitmode, and ideally just have a script bound to my arm switch, and when i arm, it takes it out of pit mode, and powers up with whatever setting is already there.

papayou66, 2017-12-03:
@wkochFPV is your script always work with new lua ? 
Thanks

spuder, 2018-02-11:
This feature would be so very helpful in the team endurance races that MultiGP is starting. 

At Sebring 2018 I watched pilots practicing the stick movements to get quads into/out of 'pit mode' as fast as possible. It would be much superior to be able to power down a vtx with 1 switch instead of 15 stick movements in team races.

niget2002, 2018-03-26:
The VTXProg script does not appear to work with lua v1.0.1 . Quick glance shows that at least the loadscript is not pointing to the correct directory for msp_sp.lua . I haven't had a chance to test to see if just changing that to point to the correct script fixes it, or if there are other changes in the latest lua too.

bseishen, 2018-04-17:
Any word on this? Could we sunset one of the 32 id's that isn't used anymore?

ghostface, 2018-08-18:
@wkochFPV i fixed your script to work with the latest version of betaflight/luascripts/opentx

https://gist.github.com/ghostface/b7de909b24fc7ce4b4c75de515c0ae46

Pascal-m, 2018-10-11:
@wkochFPV Thank you, very good script.
Is it possible to configure only the power in the script without the band and channel?
Because I use the original script "betaflight-tx-lua-scripts" for setting the frequency.
Thank you

JosefB87, 2019-09-23:
This was a very useful script, I used it quite a lot on OpenTX 2.2.2. Unfortunately after updating OpenTX to 2.2.4, (and Lua to 1.3.0, with Betaflight 4.0.5) it doesnt work anymore, I can select the script on the LUA scripts page, but nothing else happens, I can't configure any inputs. Does someone know what could be the reason and update the script so it works on newer versions on OpenTX and Lua? Or does it work on these versions and I have some other problem now?

Nikel9O, 2020-01-14:
if anyone happens to land here in search for a solution... 
here is an overkill solution: [vtx-power-on-a-switch](https://github.com/Nikel9O/vtx-power-on-a-switch)

irgusite, 2020-01-27:
If someone's still interested, I use this code (modified to work) https://gist.github.com/irgusite/1152c8b0e009cf1f6702da6c8f50464e I don't really maintain it, but I have it working so here it is :)