betaflight/betaflight #3095: f7 ANYFC STM32F745 100lqfp 216MHz MPU6000 SPI F7 (ESC FLASH)

mattyra, 2017-05-15:
Hello,

The Esc flash is not working correctly through the board, have tried flashing through classic Naze and Revo and all good same esc on the ANYFC F7 board fails, seems to corrupt the ESC data when connected to BLHELI

Any help would be good, ah yeh point to note the ESC 1 Will flash and read and Write all okay just 2,3,4 fail  

(i will fly better with a f7 right ;-) )))

mattyra, 2017-05-15:
This is the error in BLHELI 

setup failed: cmd_DeviceRead ACK_D_GENERAL_ERROR

mattyra, 2017-05-15:
2017-05-15 @ 21:33:58 -- Running - OS: MacOS, Chrome: 58.0.3029.110, Configurator: 1.0.13
2017-05-15 @ 21:34:04 -- Serial port successfully opened with ID: 16
2017-05-15 @ 21:34:04 -- MultiWii API version received - 1.31.0
2017-05-15 @ 21:34:04 -- Flight controller info, identifier: BTFL, version: 3.1.7
2017-05-15 @ 21:34:04 -- Running firmware released on: Apr 3 2017 21:52:12
2017-05-15 @ 21:34:04 -- Board: ANY7, version: 0
2017-05-15 @ 21:34:04 -- Unique device ID received - 0x3500323335511735303934
2017-05-15 @ 21:34:07 -- Reading setup
2017-05-15 @ 21:34:09 -- Reading setup finished
2017-05-15 @ 21:34:25 -- Reading setup
2017-05-15 @ 21:34:25 -- Reading setup finished
2017-05-15 @ 21:34:32 -- Reading setup
2017-05-15 @ 21:34:33 -- Reading setup finished
2017-05-15 @ 21:34:43 -- Writing setup
2017-05-15 @ 21:34:43 -- ESC 1: no changes
2017-05-15 @ 21:34:44 -- ESC 2: no changes
2017-05-15 @ 21:34:44 -- ESC 4: no changes
2017-05-15 @ 21:34:44 -- Writing setup finished
2017-05-15 @ 21:34:44 -- Reading setup
2017-05-15 @ 21:34:45 -- Reading setup finished
2017-05-15 @ 21:34:59 -- Writing setup
2017-05-15 @ 21:35:00 -- ESC 1: no changes
2017-05-15 @ 21:35:00 -- ESC 2: writing setup failed: cmd_DeviceRead ACK_D_GENERAL_ERROR
2017-05-15 @ 21:35:00 -- Writing setup finished
2017-05-15 @ 21:35:00 -- Reading setup
2017-05-15 @ 21:35:01 -- Reading setup finished

mattyra, 2017-05-15:
- Reading setup
2017-05-15 @ 21:53:10 -- Reading setup finished
2017-05-15 @ 21:53:17 -- ESC 4: flashing started
2017-05-15 @ 21:53:18 -- ESC 4: flashing failed - Error: cmd_DevicePageErase ACK_D_GENERAL_ERROR at recvCallback (chrome-extension://mejfjggmbnocnfibbibmoogocnjbcjnk/js/_4way_if.js:198:37) at chrome-extension://mejfjggmbnocnfibbibmoogocnjbcjnk/js/_4way_if.js:296:29 at Array.forEach (native) at Object.onread (chrome-extension://mejfjggmbnocnfibbibmoogocnjbcjnk/js/_4way_if.js:278:18) at read_serial (chrome-extension://mejfjggmbnocnfibbibmoogocnjbcjnk/js/serial_backend.js:234:15) at EventImpl.dispatchToListener (extensions::event_bindings:388:22) at Event.publicClassPrototype.(anonymous function) [as dispatchToListener] (extensions::utils:149:26) at EventImpl.dispatch_ (extensions::event_bindings:372:35) at dispatchArgs (extensions::event_bindings:244:26) at dispatchEvent (extensions::event_bindings:253:7)

DieHertz, 2017-05-15:
That's not a log, you should press `Save Debug Log`, not copy infprmational console output by hand.

mattyra, 2017-05-27:
[Log 2.txt](https://github.com/betaflight/betaflight/files/1033752/Log.2.txt)

mattyra, 2017-05-27:
Running - OS: <strong>MacOS</strong>, Chrome: <strong>58.0.3029.110</strong>, Configurator: <strong>1.0.14</strong>
Running - OS: <strong>MacOS</strong>, Chrome: <strong>58.0.3029.110</strong>, Configurator: <strong>1.0.14</strong>
Running - OS: <strong>MacOS</strong>, Chrome: <strong>58.0.3029.110</strong>, Configurator: <strong>1.0.14</strong>
Running - OS: <strong>MacOS</strong>, Chrome: <strong>58.0.3029.110</strong>, Configurator: <strong>1.0.14</strong>
Running - OS: <strong>MacOS</strong>, Chrome: <strong>58.0.3029.110</strong>, Configurator: <strong>1.0.14</strong>
Running - OS: <strong>MacOS</strong>, Chrome: <strong>58.0.3029.110</strong>, Configurator: <strong>1.0.14</strong>
Running - OS: <strong>MacOS</strong>, Chrome: <strong>58.0.3029.110</strong>, Configurator: <strong>1.0.14</strong>
Connecting to: /dev/tty.usbmodem1421
SERIAL: Connection opened with ID: 74, Baud: 115200
Serial port <span style="color: green">successfully</span> opened with ID: 74
MultiWii API version <span style="color: green">received</span> - <strong>1.35.0</strong>
Flight controller info, identifier: <strong>CLFL</strong>, version: <strong>2.0.2</strong>
Running firmware released on: <strong>Apr 22 2017 14:22:23</strong>
Board: <strong>ANY7</strong>, version: <strong>0</strong>
Unique device ID <span style="color: green">received</span> - <strong>0x3500323335511735303934</strong>
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_OK 0 144,243,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 14,8,21,9,9,4,255,3,255,10,3,1,1,85,170,1,255,255,255,255,255,3,255,255,255,4,250,40,80,4,255,2,0,122,255,1,1,0,3,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,68,89,83,95,88,77,50,48,65,35,32,32,32,32,32,35,66,76,72,69,76,73,35,70,51,57,48,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 0
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_OK 0 144,243,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
ESC 2 read settings failed cmd_DeviceRead ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_OK 0 144,243,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
ESC 3 read settings failed cmd_DeviceRead ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_OK 0 144,243,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
ESC 4 read settings failed cmd_DeviceRead ACK_D_GENERAL_ERROR
Reading setup <span style="color: green">finished</span>
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_OK 0 144,243,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 14,8,21,9,9,4,255,3,255,10,3,1,1,85,170,1,255,255,255,255,255,3,255,255,255,4,250,40,80,4,255,2,0,122,255,1,1,0,3,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,68,89,83,95,88,77,50,48,65,35,32,32,32,32,32,35,66,76,72,69,76,73,35,70,51,57,48,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 0
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_OK 0 144,243,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
ESC 2 read settings failed cmd_DeviceRead ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_OK 0 144,243,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
ESC 3 read settings failed cmd_DeviceRead ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 3
received cmd_DeviceInitFlash ACK_OK 0 144,243,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
ESC 4 read settings failed cmd_DeviceRead ACK_D_GENERAL_ERROR
Reading setup <span style="color: green">finished</span>
Reading setup
sending cmd_DeviceInitFlash 0 0
received cmd_DeviceInitFlash ACK_OK 0 144,243,100,1
[... 657 lines cut ...]
received cmd_DeviceInitFlash ACK_OK 0 144,243,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_OK 1a00 14,8,21,9,9,4,255,3,255,10,3,1,1,85,170,1,255,255,255,255,255,3,255,255,255,4,250,40,80,4,255,2,0,122,255,1,1,0,3,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,68,89,83,95,88,77,50,48,65,35,32,32,32,32,32,35,66,76,72,69,76,73,35,70,51,57,48,35,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32,32
sending cmd_DeviceReset 0 0
received cmd_DeviceReset ACK_OK 0 0
sending cmd_DeviceInitFlash 0 1
received cmd_DeviceInitFlash ACK_OK 0 144,243,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
ESC 2 read settings failed cmd_DeviceRead ACK_D_GENERAL_ERROR
sending cmd_DeviceInitFlash 0 2
received cmd_DeviceInitFlash ACK_OK 0 144,243,100,1
sending cmd_DeviceRead 1a00 112
received cmd_DeviceRead ACK_D_GENERAL_ERROR 1a00 0
[... cut ...]
received cmd_DeviceRead ACK_OK 1a00 14,8,21,9,9,4,255,3,255,10,3,2,1,85,170,1,255,255,255,255,255,3,255,255,255,5,255,40,80,4,255,2,0,122,255,1,1,0,3,0,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,35,68,89,83,95,88,77,50,48,65,35,32,32,32,32,32,35,66,76,72,69,76,73,35,70,51,57,48,35,32,32,32,42,42,70,76,65,83,72,42,70,65,73,76,69,68,42,42
sending cmd_DeviceReset 0 3
received cmd_DeviceReset ACK_OK 0 0
Reading setup <span style="color: green">finished</span>

mattyra, 2017-05-27:
@DieHertz see update log.....

mattyra, 2017-06-06:
any help guys ??

closed 2018-04-21 by DieHertz

jhames05, 2023-08-06:
Error: cmd_DevicePageErase ACK_D_GENERAL_ERROR at recvCallback (chrome-extension://fhkegnhbapdgoaodljlmilgakkdnlhpb/js/_4way_if.js:200:37) at chrome-extension://fhkegnhbapdgoaodljlmilgakkdnlhpb/js/_4way_if.js:298:29 at Array.forEach () at Object.onread (chrome-extension://fhkegnhbapdgoaodljlmilgakkdnlhpb/js/_4way_if.js:280:18) at read_serial (chrome-extension://fhkegnhbapdgoaodljlmilgakkdnlhpb/js/serial_backend.js:234:15)

same with mine hayssss