betaflight/betaflight #3096: Quad spins like crazy after roll

christianmccabe, 2017-05-15:
After putting in a new FC (the other died for no apparent reason) I got issues straight away when putting in full stick for a roll.
After becoming level the quads flips out like crazy. In the logfile it looks like motor 3 stays high.

At first I thought of a desync. Checked blheli and indeed motor 4 (?) showed an inconsistent PPM min throttle value. Adjusted. But when calibrating in the motors tab in BF this inconsistent motor 4 value returns.
Also thought of a gyro overflow error, but also when dropping the gyro update speed to 4khz nothing is solved.

**Setup:**
SP Racing F3 EVO (FW: # Betaflight / SPRACINGF3EVO 3.1.7 Apr  3 2017)
Littlebee 30A-S (FW: 16.6 A-H-15 Multi)
Emax red bottom RS2205 2300kv

**Logfile:**
www.mccabe.nl/zooi/LOG00001.BFL

Looks very similar to this issue, which doesn't show a real fix other than downgrading BF.
https://github.com/betaflight/betaflight/issues/430

brucesdad13, 2017-05-15:
That has happened to me when I forgot to set yaw_motors_reversed = ON. Might be totally irrelevant but triple check your prop direction?

christianmccabe, 2017-05-15:
Not relevant, the quad flies. It comes off the ground and all is fine until sharp stick input on roll (look at the log file). Motor direction and prop orientation is fine, don't worry. :)

Cheredanine, 2017-05-15:
motor 3 desyncs, possibly need to up min throttle but to be honest you go hyper on the roll stick,possibly too much tension on the gimbal? getting rid of that would help

Not a software problem

Vidalcris, 2017-05-16:
I was searching the cause of the same problem on my quad ...
Even after setting the min throttle higher the problem was still here.
In fact after one weeks of research that was the min throttle ...

I'm using Dshot600 so my first modification was from the stock value 4.5 to 5.. that was not enough ;)
Today with the min throttle at 7% i have not seen the problem happening again even with crazy full right then full left roll !

What i mean is that we should always try to put min throttle way higher in case of death roll/desync problems.

Cheredanine, 2017-05-16:
Thing is we want digital idle or min throttle as low as possible so when we are inverted we are not zooming into the ground, so it becomes a balancing act.
Finding with good motors that are not overproped I can lower stock and not have issue btw

Vidalcris, 2017-05-16:
Sure we dont want it too high, but i mean we should try to set it pretty high in case we are not sure. Then if the problem is solved we can search the perfect balance like you said.
Just to not lose time to search another source to this problem ;)

Cheredanine, 2017-05-16:
Not really, it isn't that we don't want it too high, we want it as low as we can get it.
Generally I start with digital idle at stock or even 4.0 and reduce it.
But as I say, depends on your gear

Vidalcris, 2017-05-16:
I think my bad english is the problem ;)
Anyway, i totally agree with u.

ABLomas, 2017-05-18:
Maybe looks similar? But at least on mine motors do not go full throttle:
https://www.rcgroups.com/forums/showpost.php?p=37508398&postcount=2928
I disabled accelerometer (which was not calibrated) and did one flip, it did not go into endless spin, but more testing required (not possible ATM).

Vidalcris, 2017-05-18:
Man, increase min throttle, this is exactly the same problem imo.
When u do a fast roll two motor can reverse because of the wind generated by the roll itself.
This is why the problem only happen during fast rolls (or pitch) and not on any full throttle ;)

ABLomas, 2017-05-18:
Not even close - i think you did not bothered to check blackbox logs...
Also i did later test with much higher digital_idle - does the same, so problem is def. not here.
OFC this does not mean that OP issue is not with min_throttle.

Vidalcris, 2017-05-18:
If u already tried so yeah this is maybe not that. i hope u tried high enough ;)

BlackJohnny, 2017-05-20:
I have found this issue because I was looking for a similar behavior. My quad uses Thyphoon 30A with DSHOT600, Omnibusf4 Pro, T-motor F40 2600KV and in my case the drone was flies ok for about 1 minute and then suddenly rolls back always on the same side (describing a parabola path). Because of this I started to change motors but I did not fix the issue, it did the same but on the other side. Eventually I have ended up changing all the motors but without fixing the problem. After all these I decided to try on DSHOT 300 and this was a somewhat good decision. The issue lessened (not happening every flight) however often you would hear some weird motor desync sounds and the drone would change for a second its fly attitude.

Now the weird part .... I was always logging the flight and always the log stopped just before the incident so I could not see any anomaly in the logs. I suspected that due to the motor the FC reboots or get frozen.

Investigating the backlog settings I have noticed that it was set to 100%. Thus, I have changed it to 25% (4khz) and this fixed the behavior entirely. Being a new drone I didn't have the will to go back on DSHOT600 and test. I was lucky to not brake a lot during these crashes and I wanted to enjoy it for a while ... eventually I will test it with DSHOT 600.

**And one important note: during my initial flights although I had the card in the drone, I was not using the logging because I did configure any switch to start it. So basically the drone did this even without logging.**

ABLomas, 2017-05-25:
Calibrated acc and... Problems gone.
I remember some ancient bug - if acc is enabled, but not calibrated (not used, mode may be not set, but acc hardware enabled) - some random reboots in air.
Flipped/rolled 3516816918851, no more reboots in air.

djymdmannen, 2017-08-09:
@christianmccabe Did you ever solve this? I looked at your BB log and we seem to have the exact same problem. 

- First a quick roll, then about 5-700ms later it goes into a spin.
- Both our gyro[roll] values reach and saturates at 2000deg/sec (mine at 2002deg/sec)
-  both our motor 3 works at 100%.
- our pid_p[roll] values increases to and stays at 281%.

This quad is running Betaflight F3 with BF 3.1.7. Apart from this issue it flies just fine. 

https://drive.google.com/open?id=0Bxt0dvXPWzdlNTVmeTc1dmpxaDQ

At 00:25.472 I made a quick roll and at 00:25.930 it goes into the roll of death.

christianmccabe, 2017-08-09:
I swapped nearly all, swapping the motor solved it for me but never managed to find any defects on the motor itself. Also bench tested it of the quad using scope etc. nothing abnormal. But switching it did remove the problem so one would say the two would be related...

closed 2017-08-09 by christianmccabe

djymdmannen, 2017-08-09:
@christianmccabe So you swapped one or more motors for new ones? and the problem went away? That's bad news for me then, since I already have replaced all motors. :)

maciek01, 2017-08-24:
the same flip of death issue with BF 3.2 on BFF3 board in QAV250. Swapped motors to brand new ones didint help.
i also had similar issue on BF 3.1.7 on rolls. This got fixed by raising min throttle to 1100.
with BF32 the issue is on spins and upping min throttle  doesn't help anymore.

I have another quad (realacc 210)in the same configuration - no problems at all.

djymdmannen, 2017-08-24:
@maciek01 The reason in my case was the 4in1 ESC. The ESC for motor 3 suddenly burst into flames.  I know ESC's from all brands can sometimes break, but this particular model seems to have serious quality issues. I guess you get what you pay for, never again Racerstar.

zukasmichael, 2018-05-15:
how do you raise min throttle on dshot 600 ?

mikeller, 2018-05-15:
@zukasmichael: Depends on what you mean by 'min throttle'.

zukasmichael, 2018-05-15:
I mean min throttle by min throttle

mikeller, 2018-05-15:
@zukasmichael: `dshot_idle_value`, set in 1/10000 of the maximum throttle value.