betaflight/betaflight #3103: OSD Tuning and options.

mixMugz, 2017-05-16:
Hi there. I have a bunch of ideas, how make OSD better.

1. Make 'flight mode' indicator like in mwOSD. **[ acro ]**.
2. Animate battery indicator depend of voltage
3. Add option to show/hide battery indicator (maybe change graphics to symbols)
4. Please use separated symbols for Volts 'v' and Ampers 'a' behind numbers (instead default symbols). It's already in characted map: 7 for 'v' and 155 for 'a'
5. Make option for changing white and black (separate!!!) color opacity in OSD.

jflyper, 2017-05-17:
>Hi there. I have a bunch of ideas, how make OSD better.
>
>1. Make 'flight mode' indicator like in mwOSD. [ acro ].

There is a issue thread somewhere asking this...

>2. Animate battery indicator depend of voltage

https://github.com/betaflight/betaflight/pull/3004

>3. Add option to show/hide battery indicator (maybe change graphics to symbols)

You want to do this in-flight?

>4. Please use separated symbols for Volts 'v' and Ampers 'a' behind numbers (instead default symbols). It's already in characted map: 7 for 'v' and 155 for 'a'
>5. Make option for changing white and black (separate!!!) color opacity in OSD.

How do you want use/switch them?

DanNixon, 2017-05-17:
> Make option for changing white and black (separate!!!) color opacity in OSD.

What do you actually mean by that?
Do you mean toggle drawing of white and black pixels?
IIRC that probably isn't possible as they are defined in the font.

@jflyper Did you figure out how to use the additional font memory on the AB7456?
If so we should really shove the logo in the extended memory and keep the "main" memory the same map as MWOSD, then we could at least use the correct symbols, etc.

jflyper, 2017-05-17:
@DanNixon 
>@jflyper Did you figure out how to use the additional font memory on the AB7456?

Yes, I've got a piece of paper (pdf) in Chinese and (I believe I have) decrypted it.
Haven't tried it though.

> If so we should really shove the logo in the extended memory and keep the "main" memory the same map as MWOSD, then we could at least use the correct symbols, etc.

Yeah, good idea.

mixMugz, 2017-05-17:
Guys. Give me more time. I make picture.

isilcala, 2017-05-18:
@jflyper I'm Chinese, happy to help you with the pdf if you need ;)

jflyper, 2017-05-18:
Oh, wait... I've lost the pdf... where did it go???