betaflight/betaflight #3108: AirMode set in GUI prevents correct Fail Safe operation

waltr1, 2017-05-17:
V3.1.7- this is reported by Woody_99 in Boris' thread:
Initial incident:
I hit a gate, quad crashed to the ground, disarmed and the motors kept spinning at a high rate.
Upside down on the ground the quads trying to dig a hole, and no response from the transmitter.
After the battery ran down, and burned up a motor, I found that the receiver plug had been knocked loose in the initial impact, hence no rc control.

Why did it not go to failsafe when the receiver became disconnected? 

Bench testing results:
What I did do was turned off air mode in the gui, put it back on a switch. Set motors not to spin when armed (my usual default).
Now if I unplug the receiver, it goes in to failsafe and shuts off the motors. The beeper is going, and I think it was going during the crash too, as I would expect.

Seems like airmode on in the gui overrides failsafe, at least on first experiment.

Edit:
Experimented a little bit more, and repeatedly airmode being on in the GUI caused the motors to continue to spin when I disconnected the rc receiver.
Also tried turning off the RC transmitter, and with airmode on in the GUI, the motors continued to spin as well.
Turn off airmode in the GUI and failsafe works as expected.

I'm going back to the way I had it before, with airmode on a switch. That little episode cost me a new motor and scared the crap out of me in the process.

lostcontrol, 2017-05-18:
After seeing this on rcgroups yesterday, I tried it myself on my setup. I have airmode always on, motor stop is disabled, Frsky XSR on SBUS. If I turn off my Taranis while armed (and motors spinning), the quad correctly go into failsafe.

borisbstyle, 2017-05-18:
This means that your failsafe of your receiver is not set correctly. Your receiver needs to indicate the FC that there is an actual failsafe.

On frsky products this typically means you have to set your receiver failsafe to "no pulses"

waltr1, 2017-05-18:
Boris,
Why did it not go to failsafe when the receiver became disconnected?

This has NOTHING to do with the RX setup for NO Pulses!!

Woody9999, 2017-05-18:
I believe that the point is being missed that the receiver was completely disconnected, therefore receiver failsafe setting is not a factor. 
This is on Spektrum if that makes a difference.
3.17 on BF board
I was able to reproduce the issue each time, by unplugging the receiver, motors would NOT stop spinning.
Also able to reproduce it leaving the receiver connected, and turning off the transmitter.

Turning airmode off in the GUI, failsafe works as expected, both by unplugging the receiver or turning off the transmitter. Further evidence that failsafe in the receiver is set correctly.

Please take another look.

mikeller, 2017-05-19:
@lostcontrol: Did you make sure you had throttled up enough to go over `airmode_activate_throttle` in order for airmode to become active when you did your tests? I would imagine that if this is a problem, it will only become apparent once airmode has been activated.

borisbstyle, 2017-05-19:
Try again and you will see that it simply works ;)

Woody9999, 2017-05-19:
I guess no point beating a dead horse, but I'm sorry to say that it does not work in the configuration I described. 
;)

Woody9999, 2017-05-20:
Attached is a very short video, with airmode turned on in the gui.

[Failsafe.zip](https://github.com/betaflight/betaflight/files/1015751/Failsafe.zip)

mikeller, 2017-05-20:
Looks like failsafe is recognised by the flight controller (beeping), but the motors keep spinning.

fftunes, 2017-05-20:
The first 2 answers indicate they didn't even read the initial report...

lostcontrol, 2017-05-20:
@fftunes Could you please explain how my answer makes you think I didn't read the initial report??? Cause I did, both here and on rcgroups.

@mikeller No, indeed I didn't. Just armed. I'll give it a try tomorrow.

Woody9999, 2017-05-20:
Please note that this is using Spektrum tx/rx, not that it should matter, but it may.
When airmode is on in the gui, the motors spin as soon as quad is armed. I expected the motors not to spin until airmode threshold reached.
No level 2 failsafe settings, only level 1, set to disarm and start beeper. 

All works perfectly until I enable airmode in gui, then motors will not stop if the quad has been armed and failsafe initiated.

fftunes, 2017-05-20:
@lostcontrol because the initial report was about the rx being physically disconnected.

Woody9999, 2017-05-21:
Would it be OK if maybe the folks here that know the code could look at what happened, and how to take care of it??
If it's something I did, that's fine, just let me know. I just don't want someone else to have the same issue if it can be avoided.
Flying quads is supposed to be fun, and I for one, really appreciate all the effort that's gone in to BF.

lostcontrol, 2017-05-21:
@fftunes Yes, sure, but if you read the report he says:
> Also tried turning off the RC transmitter, and with airmode on in the GUI, the motors continued to spin as well.
So turning off the TX seems to be another way to reproduce the issue. This is what I tried.

fftunes, 2017-05-21:
@Woody9999 i'm sorry if i sounded that way. I find this to be an important issue, and the way it was immediately put down annoyed me, to say the least. You were lucky it didn't fly away and "only" burn gear, I wouldn't want to imagine what could happen if this for example happened after a crash in a race, with a quad going taz into the crowd. I would appreciate a lot if the developers took safety issues more serious.

@lostcontrol It is still completely different from physically unplugging the RX.

waltr1, 2017-05-21:
I just tried duplicating Woody's issue:
Tried V3.1.5 & 3.1.7
MOTOLAB target on a Cyclone FC
X4R-SB RX - Sbus
Tried with DSHOT300 and OneShot125
This is a spare FC and used an O'scope on Motor output #1.

Whether I turned off the TX or PULLED the RX connector it always went into FailSafe - The motor outputs went to min_command.

This may be an issue with Spectrum radios/RXs. Early releases of 3.1.x did have Spectrum bugs.
We currently need some other people that use Spectrum to try this and verify if there is an issue.

Woody9999, 2017-05-21:
One other variable is I'm using BF board and firmware. 

I do have a couple of different brand FC's with Spektrum, but they are on 3.16.

mikeller, 2017-05-21:
@waltr1: Thanks heaps for looking into this. It looks like there is some configuration setting that is making this appear. Not sure if it's Spektrum related, the failsafe code is putting a layer of isolation between the RX code and the mixer code that controls the motor output, and judging from the @Woody9999's video, failsafe is recognised (i.e. it starts beeping), but the motors keep spinning.

Can you post a `diff` of your configs, @Woody9999, @waltr1?

Woody9999, 2017-05-21:
Diff with airmode on in gui.
CLI

# diff
# Betaflight / BETAFLIGHTF3 3.1.7 Apr  3 2017 / 21:43:25 (e1c4b5c)
name Chameleon

feature -VBAT
feature -FAILSAFE
feature -TELEMETRY
feature -CURRENT_METER
feature -OSD
feature MOTOR_STOP
feature AIRMODE
map TAER1234
aux 0 0 0 1700 2100
aux 1 28 1 1325 2100
aux 2 13 2 900 1300
aux 3 26 1 1700 2100
adjrange 0 0 3 1675 2100 15 4
adjrange 1 0 3 1325 1675 18 4
adjrange 2 0 3 900 1325 8 4
rxfail 4 s 1000
rxfail 5 s 1000
rxfail 6 s 1000
set min_check = 1020
set rc_interp = MANUAL
set rc_interp_int = 11
set motor_pwm_protocol = DSHOT300
set serialrx_provider = SPEK2048
set vbat_scale = 112
set vbat_min_cell_voltage = 32
set ibat_scale = 265
set acc_hardware = NONE
set pid_process_denom = 2
set blackbox_rate_denom = 4
set osd_time_alarm = 3
set osd_vbat_pos = 2456
set osd_flytimer_pos = 406
set osd_ontimer_pos = 374
set osd_flymode_pos = 387
set osd_horizon_pos = 200
set osd_mah_drawn_pos = 2435
set osd_craft_name_pos = 393
set osd_pid_roll_pos = 2305
set osd_pid_pitch_pos = 2369
set vcd_video_system = 2
profile 0

set d_notch_hz = 184
set p_pitch = 55
set d_pitch = 38
set d_roll = 33
set p_yaw = 80
rateprofile 0
rateprofile 0

set rc_rate = 105
set rc_rate_yaw = 105
set rc_expo = 10
set roll_srate = 78
set pitch_srate = 78
set yaw_srate = 78

#

Woody9999, 2017-05-21:
Diff with airmode off in gui.

CLI

# diff
# Betaflight / BETAFLIGHTF3 3.1.7 Apr  3 2017 / 21:43:25 (e1c4b5c)
name Chameleon

feature -VBAT
feature -FAILSAFE
feature -TELEMETRY
feature -CURRENT_METER
feature -OSD
feature MOTOR_STOP
map TAER1234
aux 0 0 0 1700 2100
aux 1 28 1 1325 2100
aux 2 13 2 900 1300
aux 3 26 1 1700 2100
adjrange 0 0 3 1675 2100 15 4
adjrange 1 0 3 1325 1675 18 4
adjrange 2 0 3 900 1325 8 4
rxfail 4 s 1000
rxfail 5 s 1000
rxfail 6 s 1000
set min_check = 1020
set rc_interp = MANUAL
set rc_interp_int = 11
set motor_pwm_protocol = DSHOT300
set serialrx_provider = SPEK2048
set vbat_scale = 112
set vbat_min_cell_voltage = 32
set ibat_scale = 265
set acc_hardware = NONE
set pid_process_denom = 2
set blackbox_rate_denom = 4
set osd_time_alarm = 3
set osd_vbat_pos = 2456
set osd_flytimer_pos = 406
set osd_ontimer_pos = 374
set osd_flymode_pos = 387
set osd_horizon_pos = 200
set osd_mah_drawn_pos = 2435
set osd_craft_name_pos = 393
set osd_pid_roll_pos = 2305
set osd_pid_pitch_pos = 2369
set vcd_video_system = 2
profile 0

set d_notch_hz = 184
set p_pitch = 55
set d_pitch = 38
set d_roll = 33
set p_yaw = 80
rateprofile 0
rateprofile 0

set rc_rate = 105
set rc_rate_yaw = 105
set rc_expo = 10
set roll_srate = 78
set pitch_srate = 78
set yaw_srate = 78

#

mikeller, 2017-05-21:
Thanks @Woody9999: Looking at the diffs, one thing stands out:
```
set rc_interp = MANUAL
set rc_interp_int = 11
```

Woody9999, 2017-05-21:
I can change that if you wish. 

I just tried another quad, motolabs cyclone BF version 3.15. It also exhibited the same behavior if I turn on airmode in gui. Motors continue to run.

mikeller, 2017-05-21:
@Woody9999: Yes please. At this stage we are trying to isolate what conditions cause the problem.

Woody9999, 2017-05-22:
Not sure if the edit above was noticed, but I did set rc_interp to auto. 
No change, motors continue to run in failsafe condition.

mikeller, 2017-05-22:
:-( There goes one easy to trace possibility. Thanks for testing.

Woody9999, 2017-05-22:
If there is any more information needed, I will be happy to supply what I can.
- 2 different quads
- 2 different flight controllers
- 2 different versions of firmware. 3.15 and 3.17

Commonality,
- both using specktrum protocol sats (actual Lemon brand sats)
- both dsmx, 2048
- both keep motors running if airmode enabled in gui, motors armed and tx turned off or rx disconnected
- Only level 1 failsafe used

I'm not sure what else to check or do, and have decided to keep airmode on a switch, which works perfectly fine.

waltr1, 2017-06-04:
kevinpratt823 Reports:
"I have no issue at all with failsafe on my lemon diversity DSMX sat quads on 3.1.7 with airmode enabled as a feature."

Woody9999, 2017-06-05:
I wish I knew what the reason is, but I can repeat the issue at any time by enabling airmode in the GUI.
Plus, I can repeat it on two different quads, with two different flight controllers.

Johnnygeek, 2017-07-17:
never hurts to start new 
reflash BF
rebind rx
test

sabshire, 2017-07-17:
Has anyone tried with the motor stop setting he is using (not spinning on arm, which is not the norm).  Could it be related somehow?

wind0r, 2017-07-18:
the diff shows that you have failsafe stage 2 disabled.
maybe stage 1 is broken and no one noticed since every person has stage 2 enabled?

add1ct3dd, 2017-07-18:
So what setups does this ACTUALLY affect? Lots of people saying not to use 3.1.7 because of it, but not real any solid evidence in here as to what causes it. I've done the above (powering on without transmitter on) lots of times no issues. Does this affect certain receiver protocols or what?

Woody9999, 2017-07-18:
Please go back and read my original and follow up comments. They have specific details on the symptom and my settings. 
This has also been repeated on two quads, and after new flash of firmware. 
Video also submitted above. 

It may well be caused by my particular configuration settings, but I don think the goal is to have air mode stay on in fail safe.

wind0r, 2017-07-18:
looks like airmode stays enabled in stage 1. that should be changed. but this is that way for a long time since noone disabled stag2.

what is your reason that you disabled failsafe stage2? your diff shows you disabled it.
in stage 2 airmode get disabled. i can not think about any reason to disable stage2

Woody9999, 2017-07-18:
I disabled stage 2 since I wanted the quad to just drop in failsafe, and didn't see anything in stage 2 I needed.

add1ct3dd, 2017-07-18:
But that is exactly what Stage 2 does if you set Drop?

Looks to me like there is an issue, yes, but it is far from what was everyone freaking out over (and the main issue is incorrect user configuration).

Woody9999, 2017-07-18:
Incorrect user configuration??
I just disabled a setting I don't need. 

If air mode on in GUI and stage 2 failsafe off don't work together (which I do not know is true), then maybe the settings should be exclusive.

add1ct3dd, 2017-07-18:
You said you wanted it to drop, Stage 2 does exactly that by default ? :) if it ain't broke don't fix it kind of springs to mind.

It's good that bugs like this are found, but imo it is also a demonstration of looking too hard into configuration stuff - it drops by default and works, which is why it's been reported so few times.

Anyway I'm out, glad it's been discovered, soon resolved, and can be fixed easily by setting it (imo) correctly.

Woody9999, 2017-07-18:
Feels like assumptions are being made about stage 2, without verifying that is the root cause.

sabshire, 2017-07-18:
Feels like people should read: https://github.com/martinbudden/betaflight/blob/master/docs/Failsafe.md

wind0r, 2017-07-18:
the root cause is that you expect stage 1 to do the stage 2 work..

stage 1 is only for a short outtage. the quad will work as setup and you hope that rx link comes back.
stage 2 is then after x,x sec to drop or land the quad..
if you disable stage 2 it is your own fault! everything workes like setup

Woody9999, 2017-07-18:
Please go back and read the information I posted and then let me know. 
If disabling mode 2 is an issue, then make it where mode 2 can't be disabled. 
Mode 1 is all I want and need for my racing quads. 
I can't deal with a failsafe condition causing the quad to spin the motors uncontrollably.

sabshire, 2017-07-18:
I find it amusing that you are so adamant about this.  You've done two things that are quite unsafe..  You have motor stop enabled (I really don't understand why this is even allowed to be honest), and you have disabled part of the failsafe.  What is your issue with enabling stage 2?

DanNixon, 2017-07-18:
@Woody9999 You definitely DO need stage 2 configured, this is the stage at which the motors are stopped and the quad falls (assuming the drop procedure is set).

Stage 1 only attempts to maintain the current attitude in the hope that the RX link comes back after a short period of time (i.e. in stage 1 the failure is potentially recoverable).

Stage 2 is activated when it is determined that the failure is irrecoverable (by a given time having elapsed with no recover of RX signal) and the craft can either attempt to land (using the land procedure) or stop all motors (using the drop procedure).

Yes, the firmware should not allow you to disable stage 2 failsafe, there is no logical reason to do this. However I'm not sure what led you to disable it, the default settings would have done exactly what you wanted (this is why this tab is hidden outside of "Expert Mode" it rarely needs to be changed).

DanNixon, 2017-07-19:
Ok, so here is what I hope to be a fairly complete description of why this specific setup caused the issues you are facing.

When you enable MOTOR_STOP and disable failsafe stage 2 and you have an unrecoverable RX loss (e.g. TX fault, RX disconnected, etc) your craft immediately enters failsafe stage 1.

In stage 1 the Y/P/R inputs are kept centered and the throttle is kept low. You also have your airmode aux channel failsafe set to disable airmode by setting it to a value outside of the airmode box range. Because of your this will result in the motors stopping (because you have MOTOR_STOP enabled).

This led to the false assumption that your failsafe configuration is OK, when in fact the craft never entered failsafe stage 2 and remained armed.

When you enable airmode (via a feature or switch) you override MOTOR_STOP, i.e. with airmode enabled your motors will spin using the throttle value defined by min_command at zero throttle.

Also when airmode is set as a feature it is not disabled when entering failsafe stage 1, therefore MOTOR_STOP has no effect and your motors continue to spin at min_command, they will NOT stop unless the RX link is restored or the battery is disconnected.

Had failsafe stage 2 been enabled using the drop procedure the craft would be disarmed after the stage 1 timeout and would be completely safe, the motors would stop and not restart until the failure was corrected and the craft was rearmed.

I hope this is enough to put this issue to bed now. As long as failsafe is correctly configured there is no safety issue with using airmode as a feature.

brucesdad13, 2017-07-19:
Great explanation. Thanks @DanNixon 👍

wind0r, 2017-07-19:
@Woody9999  Please go back and read the information people posted!
Do not be that Ignorant and belief what people tell you. You want Stage 2 enabled.

How you can insinuate that we did not read your post. Failsafe is under Expert Settings. You may not want to toggle that switch!

You can disable Stage 2 since for Fix Wings and other builds that are not Quads Stage 2 isnt that what people want sometimes. Holding Flaps at a defined Position

But since you are a special Snowflake and we care about you with 3.2 Stage 2 can not be disabled anymore!

I hope you are happy now.

Woody9999, 2017-07-19:
Now you're just being condescending and disrespectful. 
I point out a situation that needs attention and you basically call me ignorant.

nsccap, 2017-07-25:
I don't see how the current information on how failsafe stage1 and stage2 in the configurator failsafe tab could be improved to make this more clear, can you @Woody9999 help?

It currently says that stage1 happens at [list of conditions] and results in [set of channel configurations] being maintained/applied. Then after guard time it will enter stage2. Which has drop or land options.

I'm genuinely curious how you came to the conclusion that disabling stage2 would give you instant drop behavior?

DanNixon, 2017-07-25:
@Woody9999 You did disable failsafe, stage 1 and stage 2 are not two separate failsafe systems, they are part of a single failsafe mechanism. More importantly stage 2 is the part that actually makes the quad fail in a safe manner.

While the configurator help may not be perfect, I really cannot see why you thought disabling stage 2 was the right thing to do.

Woody9999, 2017-07-25:
If stage 2 is a requirement, then if you would like my input, I would suggest that there is no option to turn it off.
To me, stage 1 is like receiver based failsafe, where when signal is lost the channels go to a predetermined position. That's really all I need, and if I recall correctly, is what we had for failsafe in the past before introduction of stage 2, right?

It would seem that not everyone is clear about the dependency that stage 2 is enabled, since I made the inquiry about what may have happened 2 months ago and just received a clear explanation a few days ago. (thank you DanNixon) 
Unfortunately, my inquiry about what may have happened, and what settings I may have had incorrect were construed to calling it a bug and then the hype train took over. 

At the risk of being judged again, I do have one more clarifying question about stage 1... 
I had stage 1 set to disarm, turn off airmode (when on a switch) and turn the beeper on. 
That was working fine until I put airmode on as a feature, and no longer had a switch option to turn airmode off/on.
I understand that airmode overrides motor stop, but why did the disarm function not seem to work when airmode was turned on as a feature? The beeper still worked. Wouldn't disarm override airmode?

Again, I'm not trying to belabor the point, but on the other hand, would like to understand what I'm missing in that particular configuration.

DanNixon, 2017-07-25:
> If stage 2 is a requirement, then if you would like my input, I would suggest that there is no option to turn it off.

That is a valid point. See #3562 

> That's really all I need, and if I recall correctly, is what we had for failsafe in the past before introduction of stage 2, right?

Yes and no. It is similar in operation but not equivalent. It it certainly not all you need (i.e. you definitely do need stage 2).

In stage 1 you cannot disarm via a box mode setting (which is news to me too and maybe not ideal behavior). However keep in mind that stage 1 is designed to keep the craft in the air in the hope of a RX link recovery, stage 2 is what makes the craft safe (i.e. disarms it).

> It would seem that not everyone is clear about the dependency that stage 2 is enabled

In my view everyone is very clear of this given that as long as I can remember you could disable stage 2 and this is the first time I've heard of anyone actually doing it.

> I do have one more clarifying question about stage 1...

As I've mentioned, stage 1 cannot disarm the quad (even if you set the aux channel to be out of range of the ARM box mode), therefore your quad NEVER disarmed regardless of airmode configuration, it only ever looked like it had.

Personally I never bother with any stage 1 configuration and just rely on stage 2 to disarm my quad (i.e. the Betaflight default) which is probably true for most pilots. It definitely sounds like this is what you want in which case you should just not touch the failsafe settings.

Woody9999, 2017-07-25:
@DanNixon I really do appreciate you taking the time to provide detailed feedback to my questions. Very helpful for sure.
I will also remove the fallback settings for stage 1 and just rely stage 2 to disarm and drop. 

I did notice that with stage 2 enabled, the beeper beeps at a different cadence than it does with stage 1 configured to turn on the beeper. That's a function of the system to alert to a failsafe, right?

Thanks again!

DanNixon, 2017-07-25:
@Woody9999 Yes the beep pattern indicating failsafe is different from the BEEPER mode.

Woody9999, 2017-07-26:
I did find a use for leaving the beeper fallback setting to turn on.
It provides audible alarm if the receiver has no signal, ie wrong model selected in tx or tx not on.

closed 2018-04-21 by DieHertz