betaflight/betaflight #3116: BF 3.1.7  Omnibus F4 pro v2 PPM no signal output in Betaflight GUI when enable LED_STRIP

HMFHOBBY, 2017-05-20:
BF 3.1.7  Omnibus F4 pro v2 PPM no signal output in Betaflight GUI when enable LED_STRIP
Anyone met this ???

mikeller, 2017-05-20:
Yes, there is a timer clash between PPM and LED_STRIP on the OMNIBUSF4SD target:

https://github.com/betaflight/betaflight/blob/master/src/main/target/OMNIBUSF4/target.c#L53

As a suggestion, using resource remapping to assign the LED_STRIP output to motor 5 (A1) or motor 6 (A8) should get you around this.

HMFHOBBY, 2017-05-20:
Thank you , man !

jflyper, 2017-05-20:
And this is documented in the board wiki.

https://github.com/betaflight/betaflight/wiki/Board---OMNIBUSF4

https://github.com/betaflight/betaflight/wiki/Board---OMNIBUSF4#v317-and-later-led-strip-pin-assignment-has-changed-for-omnibusf4sd-target

> Explicit pin assignment remapping.
Users who choose (or forced in PPM case) not to migrate to the new assignment have to explicitly remap the assignment to the older default using the resource CLI command below.
>
> `resource led_strip a8`

closed 2017-08-28 by jflyper