betaflight/betaflight #3117: BF 3.2 - KISS24A current reporting

halninekilo, 2017-05-20:
While ESC voltage reporting works fine, current reporting seems broken - no readings. (nightly #1282)
[https://github.com/betaflight/betaflight/issues/3050#issuecomment-302828594](https://github.com/betaflight/betaflight/issues/3050#issuecomment-302828594)

halninekilo, 2017-05-20:
ESC_DEBUG output:
[image]

dk7xe, 2017-05-31:
Same issue found with SPR F4EVO with BF3.1.7 on my Octo "[Monster](https://youtu.be/8JElWqLkNmc)" setup with KISS24A ESC. Current reading is not working while voltage is ok.

halninekilo, 2017-07-08:
Is there any more info I can provide? Really missing this feature since I am flying graphenes and I can't go back to 3.1.7 because of the Omni F7 =/

mikeller, 2017-07-09:
Just did a little test with latest 3.2.0, and all of voltage, current, and consumption work fine for me, as reported in the 'Power & Battery' tab in configurator. Are you sure you got these settings: 

```
set current_meter = ESC
set battery_meter = ESC
```

halninekilo, 2017-07-09:
Thank you so much for testing this.

Yes, diff shows:
set current_meter = ESC
set battery_meter = ESC

Is there anything else I could try like switching to another UART? I am just irritated because voltage reporting seems to work fine.

mikeller, 2017-07-09:
Switching UARTs will not help with this, since the fact that the voltage reporting works shows that the communication is all good (and the frames have a checksum, so it's not possible that they are cut off after the voltage data, before the current data).

But: When testing some new code yesterday (#3474), I noticed that one KISS ESC that was still running firmware 1.08 (f I think) did report voltage, but not current. Can you check the firmware version of your ESCs, and update to latest if required?

halninekilo, 2017-07-10:
Nailed it!

Mike, before I close this - please tell me you have Paypal, Patreon or something similar. I'd love to buy you a beer.

mikeller, 2017-07-10:
Thanks, @halninekilo. Thanks for testing, this might be helpful for others having the same problem. If you really want to part with some money, you should consider https://github.com/betaflight/betaflight/wiki#providing-feedback-and-contributing-to-this-project .

closed 2018-04-21 by DieHertz