betaflight/betaflight #3120: f4 flight controllers, unkown esc, firmware erased

moosestang, 2017-05-20:
Is this a betaflight issue or blheli issue?  This is not limited to any one FC, this has happened with my flip32 f4's and now my revolt running betaflight.  1 or more escs will not read correctly and display a U instead of the number and i can't write to them, but if i disconnect the signal wire and wire in a naze32 rev5, it reads the firmware just fine and i can flash it just fine, so it is definitely not erased or corrupt in any way, it's very specific to the f4 processor.  It's odd that it works just fine at first, but eventually it will not see 1 or more escs.  escs that have had problems are tbs 25a and dys xs30a.

DieHertz, 2017-05-20:
It's not even unique to F4, it happens on F3 as well. And not only VCP, but CP210x boards as well. No common traits.

moosestang, 2017-05-20:
I never had it happen on f3, thought it was f4 specific.  Found a use for my old naze though.

slawek89, 2017-05-25:
I had similar problem with board based on omnibus F4 and DYS XM20A ESC. One of esc cannot be program. When I change connections betwen two esc, everything is fine.

veloeb, 2017-05-27:
#2544

gordonBimig, 2017-06-06:
confirmed and subscribed :-)

closed 2018-04-21 by DieHertz