betaflight/betaflight #3121: KISSFC: Support ESC sensor on TLM pin without soldering

robertlacroix, 2017-05-21:
When switching a full KISS setup (KISSFC and KISS 24A ESCs) it's apparently necessary to solder the TLM pins to a UART. At least just enabling ESC_SENSOR, current sensor from ESC sensor and configuring it to UART3 didn't work for me.

Would it be possible to support the ESC sensor on TLM (on the KISSFC) so that switching from KISS to Betaflight just requires installing the firmware and doesn't involve soldering to get the ESC sensor to work?

I imagine the problem is similar or worse on KISSCC since there isn't a telemetry wire involved :-)

mikeller, 2017-05-21:
@robertlacroix: Good suggestion. Can you please try 

[betaflight_3.2.0_KISSFC.zip](https://github.com/betaflight/betaflight/files/1016909/betaflight_3.2.0_KISSFC.zip)

and give feedback on it? Enabling 'ESC sensor' for UART3 in the Ports tab, and then setting `set esc_sensor_halfduplex = true` and `save` should do the trick.

robertlacroix, 2017-05-21:
@mikeller awesome, that got me further. The betaflight-configurator from master now shows voltage and current from the ESCs, but the combined value stays at zero. Fuel and current telemetry values sent to the remote stay at zero as well.

[image]

mikeller, 2017-05-21:
@robertlacroix: Thanks for testing. I think the remaining issues are due to bugs in the code of the Battery tab in the configurator, as far as I can tell on-board handling of voltages and currents seems to be working ok, with ESC sensors as the source, or otherwise.

robertlacroix, 2017-05-21:
👍 I'll file another bug. It can't just be the configurator because Fuel stays at 0 mAh.

PR #3127 (2017-05-23): Added 'esc_sensor_halfduplex' setting to enable KISSFC owners to use 'TLM' pads for ESC stelemetry lines.
Fixes #3121.

Setting `esc_sensor_halfduplex` to true sets `SERIAL_BIDIR` on the port used for the ESC sensor. This allows owners of the KISSFC to use the 'TLM' pads on their board for ESC telemetry connections, which are wired to UART3 TX.

(I know 'halfduplex' is somewhat daft in this context, but for RX / telemetry, `halfduplex` is used to enable `SERIAL_BIDIR` on the port, so it's at least consistent. Suggestions for a better name for all three of them are welcome.)
closed 2017-05-23 by mikeller