betaflight/betaflight #3122: F3 Betaflight 3 in 1 Flight Controller Motor 2 not spinning

miketame, 2017-05-21:
Just got new F3 board.   All wires appear hooked up right but Motor 2 is not spinning.
There is no issue with the actual motor or ESC but don't seem to be getting a signal from the flight board.
Is this a fault with the board?  Or an issue with Beta Flight?  Any suggestions??
Thank you

jflyper, 2017-05-21:
GitHub is for support and questions around development of Betaflight. For configuration questions, please go to https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread

waltr1, 2017-05-21:
Even better to go to the Thread that is Specific to the FC board for Setup and connection Details.
Do READ Wiki and check on FC Target pages for details and links.

KreativeLabs, 2017-10-21:
@miketame I have the same issue, and when I measure the esc control signal, on the other motors I have 0,77V, and on the M2 I got 3,33V. Have you resolved your issue? Thanks!

waltr1, 2017-10-21:
A DVM is not the correct instrument to use to determine if a switching waveform (PWM, OS, DS, etc signal) is properly present. Use an O'scope, even to low cost PC Sound card O'scope adapter and software works for this.

KreativeLabs, 2017-10-22:
Thanks @waltr1 for the info! I just used the DVM to compare somehow the four motors' signal. BTW I solved the issue, the board just needed an upgrade from 3.1 to the actual firmware, because the older version was buggy. Not it spins all motors, and we are happy :) Thanks!

closed 2017-10-22 by mikeller