betaflight/betaflight #3123: Timer precison on BF OSD

FPVlog, 2017-05-21:
Hi guys!

I have no idea how to do this but I know it's possible.
I would like modify the way the timer is displayed in BF OSD (I'm talking about the fly timer not On timer, even if it's almost the same thing in this case). I would need the timer to display a precison better than 1 second. Lets say 0.01s would be good. Even if its 0.1s it's already way better.

I know the processor can handle this and the reason why I would like to display a more precise timer is just because I use it as a lap timer when I go home and watch the DVR. If it was more precise I could get a lap time closer to reality.

Anyway. If someone can push me in the right direction that would be great. I know how to code and everything but never did it in BF.

Cheers

FPVlog, 2017-05-21:
I'm starting digging in the code of timer.h
Not sure if it's where I can do what I want to do but hey.... I'll give it a shot ;)

FPVlog, 2017-05-21:
ok so I am now in timer.c and it looks like it is the good file to modify to get a different way to manage the timer. I really dont know what I am doing haha.

Any help would be really appreciated!! :)

FPVlog, 2017-05-21:
ok... forget the previous messages.... it looks like it's in osd.c

I'm getting there haha

FPVlog, 2017-05-21:
All good! I managed to do it. It's probably "dirty" the way I did it but it seems to be working fine.

I modified line 223 in osd.c and instead of what was written I put:
sprintf(buff + 1, "%02d:%02d:%03d", flyTime / 60, flyTime % 60, (micros() / 1000) % 1000);

Now I got the milliseconds running next to my fly time this way I can watch the DVR and calculate my lap times more accurately!!

How cool is that?

Cheers guys!

closed 2017-05-21 by FPVlog

mikeller, 2017-05-21:
Looks good man, well done on your firmware mod!

FPVlog, 2017-05-21:
Thanks mate!!

I'll do this kind of thing more often. It's so satisfying!!!

mikeller, 2017-05-21:
You should consider making this into a pull request, so that others can start using your mod as well. That's even more satisfying. ;-)

FPVlog, 2017-05-21:
hum... I'm pretty new to this. I don't really know where to do this...

FPVlog, 2017-05-21:
i'll try though

mikeller, 2017-05-21:
It's not that hard actually. If you manage to fork the project (in the web interface), clone the fork locally (with `git`), then create a branch, and make the proposed change to your branch you are almost there already.

FPVlog, 2017-05-21:
hum ok... I'm not sure I understood everything... I'll try to do it.

mikeller, 2017-05-21:
No problem, we are happy to help. :-D

DieHertz, 2017-05-21:
When you're at it, I suggest caching the `micros()` return value for one frame in the OSD as is done with seconds/millis to avoid calling `micros()` while drawing a frame.