betaflight/betaflight #3124: Problem flashing betaflight on beecore f3 evo

willback510, 2017-05-21:
I've been trying to flash a beecore f3 evo with an spracing f3 evo firmware hex. I'm having an issue where it betaflight is saying there is no response from the bootloader. I've tried bridging the boot pins, no difference

mikeller, 2017-05-21:
Did you check 'No reboot sequence'? Also, did you bridge the boot pins **while powering up the board**?

willback510, 2017-05-21:
Yes

willback510, 2017-05-21:
Ok so I know this is amazing betaflight support page but I really need some help. So the beecore comes with cleanflight, and I did a reset on it bc I couldn't get it to flash. After a bit more trying and failure I settled with cleanflight, but now when I arm it and give it throttle, the throttle is too high, and it just spins. I've checked all the props and motors, not sure what the issue is, pls help

mikeller, 2017-05-21:
GitHub is for support and questions around **development** of Betaflight. For configuration questions, please go to https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread

mizuta1951, 2017-05-23:
I had the same situation on OMNIBUS F3 AIO. Maybe your board has pre-installed firmware. In my case it had.
It seems the the bender used FTDI interface and STM flash loader to install the firmware.
In that case if they checked the firmware protection, it will not be able to put into boot loader mode by using USB.
It should be possible to re-write the firmware using FTDI. Go to STM HP and download "Flash loader Demonstrator". Connect your FC to FTDI and run  "Flash loader Demonstrator". You will be asked that the firmware is protected and it will be full erased. Just let it full erase and then select the firmware HEX you want to put in. STM flash loader will not search the internet for Betaflight firmware, so you have to download the HEX file before doing this.
Hope it will work for you.

AndersHoglund, 2017-05-23:
There are some things that needs clarifying here... 
The "STM Flash Loader Demonstrator" can ONLY be used on serial ports, via an USB-serial adapter. Needs the the appropriate driver installed for the adapter. BetaFlight Configurator flasher can be used when connected this way too.

The "STM DefuSe Demonstrator" is similar, but can only be used on MPU with USB VCP and a bootloader with support for the USB-DFU protocol, i.e. not for F1s nor any FC using Silabs USB interface. This flasher program requires the special STM USB-DFU device driver and by default Windows installs this automatically.

Unfortunately the BetaFlight Configurator flasher can NOT use this STM USB-DFU device driver, the generic "WinUSB" driver is required. This is where Zadig and ImpulseRC Driver Fixer comes in and helps replacing the device driver, over riding the automatic default installation done by Windows. This however makes the "STM DefuSe Demo" unusable. You have to choose which flash program to use, and make sure the suitable device driver is installed.

In this particular case, the beecoreF3evo is a USB VCP device and a USB-DFU device. It needs the WinUSB driver for flashing over USB to use the BFC flasher. Or an USB-serial adapter and flashing over a serial port, using either BFC Flasher or the STM FlashLoader Demo.

Most of this is covered in the Wiki: https://github.com/betaflight/betaflight/wiki/Installing-Betaflight

closed 2017-10-23 by jflyper