betaflight/betaflight #3128: Renaming some things to allow for a new digital protocol

TonyBazz, 2017-05-21:
# A - Description

I would like to rename some files and #DEFINES in the betaflight to allow for another digital Protocol to be added. Basically i would like to see the following..
#define USE_DSHOT renamed to USE_DIGITAL
pwm_output_dshot.c renamed to pwm_output_digital.c

I want to add a function to pwm_output_digital.c to allow for each pulse width to represent a nibble of data. As described here: https://www.rcgroups.com/forums/showthread.php?2879092-%2A%2AIdea%2A%2A-ProShot-Another-digital-protocol/page1

I have made the changes locally on betaflight 3.1.7 and all works well as demonstrated in the video.

# B - Steps To Repeat

-- replace this line with a list of steps to repeat your problem

# C - Video Links

https://www.youtube.com/watch?v=7EYbWa1Kzes&feature=youtu.be

# D - Forum post links

Also discussed on the betaflight thread in RCGroups

mikeller, 2017-05-21:
I think the approach that you are taking with your new protocol has potential. However, I am not sure that creating a new name is the right way to go (also, I notice that the frame format is identical to the frame format of DShot, it's only the physical encoding that's changed).

My suggestion would be to use a name that better expresses that this is a variant of the DShot protocol, like 'DShot16'.

ledvinap, 2017-05-21:
@TonyBazz : 
Some thoughts about multi-bit per pulse:

- instead of constant space between pulses use constant length for nibble
  - time between rising edges is constant,you can use it co calibrate  ESC clock (allowing more bits per pulse)
  - pulses for all ESC can start at identical time - it will be possible to update all timer channels with single DMA or even ISR (if reasonable speed is used)
  - all packets will have data-independent length, improving protocol determinism (hunting bug that is triggered by too long or too shots packets is hard)

- maybe 'equalising' pulse may be added, so that sum of all pulses is multiple of 4 (for 4 bits/ 16 pulse lengths). This will allow compensating different delay of rising and falling edges (RC filter on ESC input; optoislolator delay,  etc), improving protocol robustness.

- it may be possible for ESC to send nibble back after receiving packet, possibly by short pulse referenced to last master pulse. 4bits per packet will allow reasonable bandwidth for telemetry

martinbudden, 2017-05-22:
@TonyBazz , I think it is best to make your PR with the current naming, and then sort out the naming after. We can then have a better idea how the names interact. For example, we may wish to enable some boards with your protocol and not DSHOT or vice versa.
Also note that `USE_DIGITAL` is far too generic a name , since DIGITAL could refer to anything. At the very least it should be `USE_DIGITAL_ESC_OUTPUT` or similar.

TonyBazz, 2017-05-27:
Gents, What do you think?
Basically just added a function pointer to Dshot or ProShot depending on the motor protocol selection.
https://github.com/TonyBazz/betaflight/commit/556c4c5bcd89b848d27f89159977bf2e35ce7d2a

martinbudden, 2017-05-27:
Looks good. Note that `pwmWriteDigital` is itself called via a function pointer, so rather than adding a new function pointer, it would be better to set the function pointer for ProShot in `motorDevInit` see https://github.com/betaflight/betaflight/blob/master/src/main/drivers/pwm_output.c#L213-L257

TonyBazz, 2017-05-27:
Good recommendation. Much cleaner. :)
https://github.com/TonyBazz/betaflight/commit/11a520b41de0120ed63b7c879662386575362fef

closed 2018-04-21 by DieHertz