betaflight/betaflight #3129: BF3.2 twitches - tuning/setup error or firmware error?

DeepnineRC, 2017-05-21:
Hi,
i'm running newest master from today. I get random twitches on moderate flying.
i can feel it only on roll axis. 

running KIWIF4V2 8k/8k Dshot600

Is it tuning/setup error or is it firmware related?

[image]

[image]

[LOG00023.BFL.txt](https://github.com/betaflight/betaflight/files/1017307/LOG00023.BFL.txt)

mikeller, 2017-05-21:
Just from looking at the screenshots, the dropouts in the logs suggest that it might be logging interfering with control. Have you tried lowering your log rate and / or disabling DMA for the SD card?

qemay, 2017-05-22:
We are experiencing similar twitches on two different FC running 3.1.7. The first was SP racing F3 running 8khz oneshot 125 (~30% load). I don't have logs for that, because I don't log on that FC. And the weird thing is those went away after some time. If those twitches are a log interference then not in my case. 
And now on a new F4 ombnibus at 8khz/dshot600 we have noticed those too. But now we have noticed them in log's also. 

[image]

Could you please give a hint about disabling DMA for SD card (sorry did not do that till now, and I don't even know where to look)

mikeller, 2017-05-22:
`set sdcard_dma = off` in the CLI, then `save`.

DeepnineRC, 2017-05-24:
problem solved by disabling dma for sdcard.

mikeller, 2017-05-24:
@DeepnineRC: If it's solved, can you close the ticket please?

closed 2017-05-25 by DeepnineRC