betaflight/betaflight #3131: Softserial not working with X4R SB on ßF > 3.1.5

Wene001, 2017-05-22:
Hi

I´m using FPracing F3 Acro and a Frsky X4R SB on IO2 and Smartport connected to pin 6.
Using 3.1.5 and Smartport connected to softserial (IO2 Pin 5+6 bridged) all is working fine.
Using 3.1.6 or 3.1.7  and Smartport connected to softserial (IO2 Pin 6) Telemetry works sometimes for 2 Seconds . Sometimes telemetry comes up again.

Is this a bug?

Greetz Werner

jflyper, 2017-05-22:
The new software serial has been around for months now. It's hard to imagine it still has a subtle bug with a popular FC like SPRF3.

I suspect you have wired the SmartPort wire to a wrong pin when you changed your wiring.

Wene001, 2017-05-22:
Ive wired the smartport pin from XSR4 to IO2 Pin6 (softserial2 TX)  on my FPRacing F3.
i tried to use Softserial 1 on IO1...also tried to remap Softserial2 TX to an other pin, using the resource command. Alway with the same result.
After Board Reset the Telemetrie is working for a few seconds.
waiting one minute or so Smartport Telemetrie works for a few seconds again
Confused ?!?

daro6, 2017-05-22:
Hi Wene001,
as you can see in issue https://github.com/betaflight/betaflight/issues/2823, the softserial feature is disabled in 3.1.7 for the SPRACINGF3 target. The last working version is 3.1.5. 
Greets,
Daniel

Wene001, 2017-05-22:
Thanks
But in my case i could enable it and it works for a few seconds after Reset
Is there a chance that this feature is available in the next verions?

mikeller, 2017-05-22:
@Wene001L SOFTSERIAL was removed from this target because it did not fit into the flash size. It should be added again in the next version.

maciek01, 2017-05-25:
I have the same issue - s.port telemetry works during the startup and then stops:

BFF3
BF 3.1.6
FRYSKY everything else

connected via hardware UART3

I did not try <316

mikeller, 2017-05-26:
@maciek01: GPS support was disabled for F3 boards in 3.1.6.

monelli, 2017-05-30:
Same problem, telemetry works for a few seconds after Reset.
Solution:
set tlm_halfduplex = OFF
Tested on 3.2.0 on softserial with SPracingF3EVO and x8r.
SmartPort Telemetry.

maciek01, 2017-05-30:
@monelli is your s.port cable attached to both TX and RX or just TX?

monelli, 2017-05-30:
(>|) is a 1N4448 diode.

```
(RX)_____________(SmartPort)
                    /
(TX)__(>|)__/
```

jflyper, 2017-06-01:
I think SmartPort for non-F3 target is broken in post-3.1.7. #3190

jflyper, 2017-10-23:
It's apparently working now (3.2.x)

closed 2017-10-23 by jflyper