betaflight/betaflight #3134: Suggestion

Wtcheesecakes, 2017-05-22:
add a switch under configuration to enable reversed motor direction instead of cli

mikeller, 2017-05-22:
Will be easy to do once the parameter group based protocol is implemented.

pechovic, 2017-05-26:
Could you please update the title of this item to be something relevant/meaningful ???

kmitchel, 2017-06-23:
It would be nice if alot of CLI stuff was exposed in the configurator.

mikeller, 2017-06-23:
@kmitchel: A big part of this is making the settings available in the protocol that is used to communicate between the flight controller and the configurator, which has to be done manually. One of the next 'big steps' in Betaflight is to introduce a protocol that will see new parameters getting into the protocol automatically as soon as they are added to the firmware.

kmitchel, 2017-06-23:
@mikeller is this a 3.2 milestone, or a longer term project?

mikeller, 2017-06-23:
I think we'd put it against 3.2 originally, but if we decide to release 3.2 earlier than the projected October release date, it will probably slip into 3.3.

pechovic, 2017-07-29:
Hi guys, did you go out with this item yesterday? Seems like it's not working. I reported the issue in the betaflight-configurator project https://github.com/betaflight/betaflight-configurator/issues/573

JOSHFPV, 2017-08-13:
I have same issue! I'm using Fortini F4 firmware 3.2 RC31

DEvil0000, 2017-08-16:
Works for me on F3 in 3.2.0 RC3 but I had to reset settings and reconfigure. Config import/migration seamed to cause a issue

closed 2018-04-21 by DieHertz