betaflight/betaflight #3139: BF 3.1.7 not working for SP F3 at (Oct X) X8 mode

Wimleung, 2017-05-24:
SP F3 Standard version (36mmx36mm). Firmware BF3.1.7. Abnormal motor speed at X8 mode.
1. The motors output was constant 1070(no matther what the throttle value was) after being armed at X8 mode. Everything looked normal if set to X4 (quad X) mode.
2. After flashed the firmware several times, I got different result. Sometimes, the motor output was constant 2000(full speed) no matter what position the throttle stick was. And sometime it was constant 1070.... That was at X8 mode. Everything looked normal if set to quad 4...

Could it be hardware fault or firmware bug? Thanks.

Wimleung, 2017-05-24:
More information.

If flash BF 3.1.3 and 3.1.6, the output was OK when on the GUI, but when put to real flight the motors No. 6 and 8 never spinned. It could fly stably at "angel" mode as long as I tried to control the pitch... While sometimes the No. 8 motor could spin(No. 6 still not)  but just shaked constantly....

All these tests were with SBUS receiver and default PID.

jflyper, 2017-05-25:
@Wimleung Can we see the output from cli command `diff` under 3.1.7?

Wimleung, 2017-05-25:
@jflyper Thanks for your response. The output is,

# diff
# Betaflight / SPRACINGF3 3.1.7 Apr  3 2017 / 22:27:23 (e1c4b5c)
mixer OCTOX8

feature -RX_PPM
feature RX_SERIAL
serial 1 64 115200 57600 0 115200
aux 0 1 0 900 1200
aux 1 2 0 1300 1700
set min_throttle = 1020
set min_command = 980
set serialrx_provider = SBUS
profile 0

rateprofile 0
rateprofile 0

# 

Actually not only 3.1.7, I tested 3.01 through 3.1.6, none worked well. Some can fly quite stably with only 6 motors spinning, but sometimes 7 spinning with constant shaking(default PID)  with the same version of firmware. 

But with the same firmware, if changed to X4, everything was OK...

jflyper, 2017-05-25:
One of the devs confirmed correct esc signal output. Why not moving esc/motor 6 & 8 around to other positions to see if something changes. I would move these motor/esc pairs from under to upper and test it in QUADX4 then OCTOX8; do it  one by one, e.g., swap esc/motor 6 and esc/motor 2, then esc/motor 8 and esc/motor 4.

Wimleung, 2017-05-25:
But for 3.1.7, the motors kept a constant spinning speed (1070 or whatever you set as lowest  throttle) after arming, and didn't respond to throttle input. So it doesn't look like to be problems of special output channels. (Motors 6&8 problem is with old versions of firmware).

jflyper, 2017-05-25:
@Wimleung 
> the motors kept a constant spinning speed (1070 or whatever you set as lowest throttle) after arming, and didn't respond to throttle input

Is this with the motors tab? Or real craft?

Wimleung, 2017-05-25:
Both.

Wimleung, 2017-05-27:
@jflyper  Both.

jflyper, 2017-05-29:
@Wimleung Confirmed OCTOs have problem with 3.1.7. Devs will look into it.
(Apologies it took so long; I was away from my bench for the weekend.)

Wimleung, 2017-05-29:
@jflyper Thank you very much. Is the for Version 1.7.1 or for all versions? Thanks again.

jflyper, 2017-05-29:
@Wimleung The omission is for 3.1.7.

Wimleung, 2017-05-29:
@jflyper Thank you very much.

Wimleung, 2017-05-29:
@jflyper Do I need the omission for F4 or F7 (my F4 is shipping) for 3.1.7? Thanks.

Wimleung, 2017-05-29:
@jflyper Just tested this on a Mini X8 with SPF3 with 3.1.7, it really worked. T

Thank you so much.

jflyper, 2017-05-29:
@Wimleung That is good to hear! Have fun :)

Oh, btw, mini-x8? What's your frame?

Wimleung, 2017-05-29:
@jklyper About140mm(diagonal) with 8pcs 1104  and 3020 prop. Could be the smallest ever known.

jflyper, 2017-05-30:
@Wimleung Your original frame? I'm interesteeeeeed! :)

Wimleung, 2017-05-30:
@jflyper It seem no picture attachment option here, would you please mail to West_forever@163.com and I will respond with pictures : )

mikeller, 2017-05-30:
@Wimleung: There is a link at the bottom of the GitHub comment edit box that allows you to upload pictures.

Wimleung, 2017-05-30:
[image]
[image]
[image]

jflyper, 2017-05-30:
@Wimleung Interesting!!! :)

mikeller, 2017-05-30:
Wow, nice!

dk7xe, 2017-05-31:
Nice to see other octo builds besides my "[Monster](https://youtu.be/8JElWqLkNmc)" ;D

Wimleung, 2017-06-03:
@dk7xe Wow! Your Monster looks much bigger.

closed 2017-10-22 by jflyper