betaflight/betaflight #3140: SD card not working on AnyFC-F7 betaflight version 3.1.7

feiticeir0, 2017-05-24:
Hi ! 
I'm having an issue getting any SD cards to be initialized by Betaflight using the SD card slot on the AnyFC-F7

The card is formated with FAT32 , is class 10 and with 8GB

When in the blackbox tab in configurator, I have the onboard SD card with the message filesystem starting...

sd_info get's me this:

SD card: Manufacturer 0x27, 7782400kB, 04/2016, v3.0, 'SD08G'
Filesystem: Initializing

With a Samsung EVO 64GB SDXC I get the same message...

dump:
`# version
# Betaflight / ANYFCF7 3.1.7 Apr  3 2017 / 21:52:12 (e1c4b5c)

# name
name -

# resources
resource MOTOR 1 B08
resource MOTOR 2 A02
resource MOTOR 3 A01
resource MOTOR 4 A03
resource MOTOR 5 B05
resource MOTOR 6 A00
resource MOTOR 7 B09
resource MOTOR 8 E06
resource MOTOR 9 B03
resource MOTOR 10 B04
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B14
resource PWM 1 B14
resource PWM 2 B15
resource PWM 3 C06
resource PWM 4 C07
resource PWM 5 C09
resource PWM 6 C08
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 B05
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 D05
resource SERIAL_TX 3 D08
resource SERIAL_TX 4 C10
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 E08
resource SERIAL_TX 8 E01
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 D06
resource SERIAL_RX 3 D09
resource SERIAL_RX 4 C11
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
[... 422 lines cut ...]
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_sensitivity = 55
set level_limit = 55
rateprofile 0

# rateprofile
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 23
set rc_yaw_expo = 18
set thr_mid = 50
set thr_expo = 0
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80
set tpa_rate = 10
set tpa_breakpoint = 1650`

mikeller, 2017-05-24:
Duplicate of #3049.

@feiticeir0: Next time, before you create an issue, try searching for existing issues first. #3049 gives a whole lot of information (including a workaround) for your situation.

Also, please note that formatting SD cards with the operating system is not optimal, in particular for situations where data is then streamed onto the card (blackbox, video camera, ...). Use the SD card formatter tool, and set the options according to this: https://kb.sandisk.com/app/answers/detail/a_id/14827/~/using-sd-formatter-tool-to-restore-full-capacity-on-sdhc%2Fsdxc-cards

closed 2017-05-24 by mikeller

feiticeir0, 2017-05-24:
Hi ! Thank you for the reply. I did search, but I've missed that one. 
I don't use (or have) windows or MAC OS X, I don't have that tool available to me. 
The workaround is to buy a scandisk ?  If you wan't me to pile on the other thread, In a few days I'll have an omnibus F4 with SD Card slot.. I can test if initializing in a F4 will work on the F7.. 
 
Best regards and thank you

mikeller, 2017-05-24:
You could always use a virtual machine running windows, or ask a fellow pilot to do this 20 second job for you. :-)

And yes please, report your findings in #3049, we are still working on figuring out what is causing this.