betaflight/betaflight #3145: BF 3.2.0 - HOTT Telemetry not working

TheAngularity, 2017-05-24:
using latest bf 3.2.0 master on airbot f4 and revolt but hott telemetry is not working ... going back to bf 3.1.7 and hott telemetry is working well

mikeller, 2017-05-24:
Have you tried different serial ports, in order to determine if the problem is in the serial port (-settings) or in the telemetry code?

TheAngularity, 2017-05-24:
no ... using same port that worked on bf 3.1.7

TheAngularity, 2017-05-27:
@mikeller tested this now with all ports on a sprf3evo
but nothing ... hott telemetry not working on all ports ... back to 3.1.7 and viola it's working

mikeller, 2017-05-28:
Thanks. Looks like HOTT is broken in 3.2. then. Thanks for testing.

martinbudden, 2017-05-28:
@TheAngularity , could you do a git bisect to pinpoint the commit that caused the problem?

TheAngularity, 2017-05-28:
@martinbudden, will be hard, but i could try it
will first search the jenkins build, maybe i find a reference point

TheAngularity, 2017-05-28:
@mikeller @martinbudden - ok, have found the guilty
with http://andwho.sytes.net:8080/job/BorisB_BetaFlight/1208/ starts the problem

mikeller, 2017-05-28:
Good find. A quick look at that guy shows that amongst other things, the `tlm_halfduplex` setting is now used for HoTT. Can you try if changing this setting makes it work for you?

TheAngularity, 2017-05-28:
@mikeller `tlm_halfduplex` set to ON or OFF makes no difference

martinbudden, 2017-05-28:
Yes, it looks like @hydra 's fix inadvertently broke something.

nhadrian, 2017-08-06:
hi all, I did somne tests and with tlm_halfduplex = OFF the HOTT telemetry works.
Controller is Airbot F4 compatible, HOTT is on UART3 with proper diode.

Regards Adrian

nhadrian, 2017-08-11:
hi again, I did some further tests, with 3.2.0 RC3 and with tlm_halfduplex = OFF setting the HOTT telemetry only partially works, datas are transmitted only once after powering on. Also warnings are not working through HOTT. 
With tlm_halfduples = ON, now the SUMD receiver even don't recognized, also telemetry partially works. (in 3.2 RC2 SUMD war working independently of the tlm_halfduplex setting) :(

Could anybody please help and try to find what causing errors in 3.2?
**Or at least, if HOTT is not a target (and market) anymore, please remove the option from the pop-up menus tolet users know support is gone.** Thanks in advance.

Controller is Airbot F4 compatible, HOTT is on UART3 with proper diode.

Regards Adrian

frizzle101101, 2017-08-13:
I can confirm that telemetry is broken on my matekf405, just trying to use sport with 3.2 rc3. The sensors are seen once but then never updated. Is there a fix for the user side.

mikeller, 2017-08-13:
@frizzle101101: Are you using HoTT or S.Port? If it's S.Port, the behaviour that you are describing is indicative of a problem with the XSR RX (and probably other models) firmware. Get the latest version from FrSky, and update it.

Paradox32b, 2017-08-15:
Hi all, i use 3.2 RC3 on flip32 omnibus f4 v2.5 and HOTT telemetry on UART3 with 1N4148 diode and i have the same problem !

nhadrian, 2017-08-15:
Looks like a common issue with HoTT protocol.

Paradox32b, 2017-08-15:
Yes it's a shame...

Paradox32b, 2017-08-15:
@nhadrian "after reading #3145 I tried 'tlm_halfduplex = OFF' setting and HOTT works!"
It does not work ?

martinbudden, 2017-08-17:
@hydra , can you have a look at this. It appears your PR https://github.com/cleanflight/cleanflight/pull/2756 is what caused the break.

hydra, 2017-08-17:
I'll give it a test when I get a break form busy work chaos.

Paradox32b, 2017-08-17:
@hydra Thanks you very much and courage !

martinbudden, 2017-08-27:
@hydra , any progress on this?

hydra, 2017-08-27:
nope. ive been having a break from the code recently.  it's still in my list when i get a chance though.

Ralfde, 2017-09-09:
who of the Betaflight team could repair the HoTT Telemetrie? Does somebody need a radio and receiver for testing?

martinbudden, 2017-09-09:
@Ralfde , I could get this working if I had a radio and receiver.

Ralfde, 2017-09-09:
Dear Martin Budden,
if you would give me your address, I will send you one set.
Best regards, 
Ralf

  (#3145)

redneonrt, 2017-09-10:
I'm happy to help test if I can, I have a mz-24 radio and a g r 12 L receiver a gr 24 receiver and a gr 18 receiver,

Ralfde, 2017-09-10:
I could send test equipment, whe I would get the address where to send it by mail.

Ralfde, 2017-09-16:
Do you think, you could get the HoTT Telemetry working again in the RC6?

hydra, 2017-09-16:
i can take a look, i just need to find time...

mikeller, 2017-09-16:
@hydra: I think @martinbudden said he'd look into this.

wildflyer, 2017-09-20:
I would need the hott telemetry also very much, no other solution for low voltage alarm (beeper or osds etc) is as good as the telemetry solution. I could test it or I also could send you a receiver.
greetings
joachim

IvaAlex, 2017-10-03:
Hello guys,
i've manage to "fix" HOTT telemetry in RC6.
I took hott.c file from build 1207 (Build 1207 (24-Apr-2017 10:09:06), and with few changes, the telemetry now works !
[hott OK.txt](https://github.com/betaflight/betaflight/files/1353342/hott.OK.txt)
if i have time in the next days, i will check the hott.c from rc6 build

Paradox32b, 2017-10-05:
Hi,
Thanks for the work but it still does not work for me.

IvaAlex, 2017-10-05:
Hmmm. It should work only on hardware serial ports, not softserial. I have tested on betaflightf3 fc, receiver connected to serial port 1 and telemetry to serial port 3. What fc are u using? maybe i can compile for you

Paradox32b, 2017-10-05:
Thanks, I use an "omnibus f4 SD v3" http://www.readytoflyquads.com/flip-32-f4-omnibus-v3 on UART3 with the diode. The telemetry worked great with version 3.1.7.

IvaAlex, 2017-10-05:
So, you took my file, overwrite the original hott.c, recompile for your board and it didn't work ? Did you configure fc for serial receiver  sumd ? Set tlm_halfduplex=off ?

Paradox32b, 2017-10-05:
No, could you compile it for me please?

IvaAlex, 2017-10-05:
[betaflight_3.2.0_OMNIBUSF4.zip](https://github.com/betaflight/betaflight/files/1360455/betaflight_3.2.0_OMNIBUSF4.zip)
Try this. if it works, please confirm here if it works. maybe @hydra or @martinbudden can make the final changes

Paradox32b, 2017-10-05:
Thank you very much, it works great !

mikeller, 2017-10-06:
@IvaAlex: How about you open a pull request for your changes? It's not complicated, and we're happy to help.

IvaAlex, 2017-10-06:
I will try to modify the last hott.c file from rc6 and after i will open a pull request. Thank you !

Ralfde, 2017-10-09:
Das Build is for the normal OmnibusF4, not for the SD version. The electric air module works, but the vario (altitude) (barometer) doesn't work. Could you please make also the builds with vario HoTT Telemetrie and also the .hex files for the F3 and F4SD and F7 from Omnibus?

IvaAlex, 2017-10-09:
@Ralfde i'll post the files today, in the evening

IvaAlex, 2017-10-09:
https://drive.google.com/open?id=0B_Q-2DPakp62UkZSZFlBWDJCMzA
3.2.0 all hex files cu HOTT fix

Paradox32b, 2017-10-09:
Awesome, thank you very much.
The first firmware you have compiled for "OMNIBUSF4SD" works for telemetry but the quad does not fly!

Ralfde, 2017-10-10:
The Vario Sensor with Omnibus is still not working. Just the Electric Air Module. Could this be fixed, too, please?

IvaAlex, 2017-10-10:
@Ralfde Sorry, but no. in the far future i want to continue PierreA work and implement some PID config from our Graupner radio. like here : https://github.com/cleanflight/cleanflight/issues/260
@mikeller i'm trying to open a pull request, but i don't know what section, and how do i mark my changes ?
https://github.com/cleanflight/cleanflight/pull/2959 this is 100% wrong :)

mikeller, 2017-10-10:
Just create a fork of  betaflight, create a branch on your fork, and apply the changes to this branch. Then, after pushing, in the GitHub web interface, select your branch and click 'Create Pull Request'.

Ralfde, 2017-10-10:
PID settings will be a great feature, too.

Paradox32b, 2017-10-10:
Like this https://www.youtube.com/watch?v=5UIVe8hPpyM it would be awesome!

croghostrider, 2017-10-10:
@IvaAlex I have for you open a Pull Request. #4325

davidbitton, 2017-10-11:
I'm having an issue (#4341) w/ Mavlink. Could it be from the same source that killed HoTT?

IvaAlex, 2017-10-12:
@Paradox32b check this thread https://github.com/betaflight/betaflight/pull/4325 , please answer @hydra

commit a9b97f0c (2017-10-19): Merge pull request #4325 from croghostrider/HOTT-telemetry-fix-from-IvaAlex

HOTT telemetry fix #3145
commit ff9aa5de (2017-10-19): Merge pull request #4325 from croghostrider/HOTT-telemetry-fix-from-IvaAlex

HOTT telemetry fix #3145
closed 2017-10-19 by mikeller