betaflight/betaflight #3148: Omnibus not showing up while in bootloader mode.

slickzilla, 2017-05-25:
installed proper drivers. Stm 32 doesnt show up on zadig. device not found in device manager. can not do anything to access flight controller 
[image]

AndersHoglund, 2017-05-25:
VCP devices behaves this way when no firmware is flashed, flash is empty, and not in bootloader mode. Or a charge-onlyUSB cable is used. Bridge the bootpads and try again... You should then see the `STM32 BOOTLOADER` device. Can of course also be a broken FC...
https://github.com/betaflight/betaflight/wiki/Installing-Betaflight

SoulPictures, 2017-05-25:
Got the Same here with an Banggood OmniBus F4. Looks completely dead, green light only, blue one stays off. Out Board here seems dead. Tested multiple computers

McGiverGim, 2017-05-25:
I had the same problem but trying other other option in zadig driver worked. Have you tried other options?

jflyper, 2017-05-25:
This is a support request.

AndersHoglund, 2017-05-25:
"Other Options"? which one??
You should only need to select option "List all Devices", as described in the Wiki.
And when in bootloader mode, no leds are on except the power indicator(s). Some FCs has both 5V and 3.3v LEDS, some only 5V.

AndersHoglund, 2017-05-25:
Please use RCG and/or FB for end user support and help issues. This is the place where Betaflight Firmware issues are reported and discussed.

ahsanwtc, 2017-08-18:
@slickzilla were you able to solve the issue?

AndersHoglund, 2017-08-18:
This is not a betaflight issue, closing. FB an RCG etc are better places....

closed 2017-08-18 by AndersHoglund