betaflight/betaflight #3152: Spektrum Satellite always in bind mode

richard-scott, 2017-05-25:
I've a SPRACINGF3EVO and I'm connecting a Spektrum Satellite to it and binding via the CLI part of the Configuration GUI.

After activating Serial RX mode for UART3 in the Ports screen and then selecting "Serial-based receiver" in the Configuration screen I then run this in the CLI to initiate bind mode:

```
set spektrum_sat_bind=3
set spektrum_sat_bind_autoreset=0
save
```
I then have to disconnect power, and reconnect power to restart my FC.

This works fine, and the orange light starts to flash.
I then bind my radio ok, and the light goes solid orange... I can then go into the Receiver screen and select the correct channel map and my sticks make the correct items move.

However... If I then restart my FC it goes direct into bind mode again with the light flashing.

I have to do this to stop it:

```
set spektrum_sat_bind=3
set spektrum_sat_bind_autoreset=0
save
```
I then have to disconnect power, and reconnect power to restart my FC. Once it has restarted and the receiver is flashing I can proceed with binding.
```
set spektrum_sat_bind=0
save
```

Is that right? I'd have thought the final bind=0 would be completed automatically for me?

I started running BF 3.1.7 (Stable) and I've just compiled 3.2.0 from Git (#4cc5b9b) and tried that too, but neither release the FC from bind mode.

richard-scott, 2017-05-25:
ah. after a google I found [this](https://github.com/cleanflight/cleanflight/blob/master/docs/Spektrum%20bind.md) which says:

> If no hardware bind plug is used the spektrum_sat_bind parameter will trigger the bind process during the next hardware reset and will be automatically reset to "0" after this

AndersHoglund, 2017-05-25:
Well, you turned off `spektrum_sat_bind_autoreset`. If you had left it on, the bind process would have automatically reset it self. But since you turned off this autoreset function, you had to do it manually. Please try again with auto reset turned on.
In some cases it is needed to handle this manually, hence this option to turn off autoreset.

richard-scott, 2017-05-26:
I can't get it to work again now. As soon as I type "save" in the CLI my board resets ;-(

richard-scott, 2017-05-26:
lol, oh yeah that works much better.

If anyone else gets this issue it could be because you followed the same page as me:
http://www.propwashed.com/spektrum-receivers-with-betaflight/

Poor information there.

One thing I have noticed is that the variable name to disable autoreset has changed between BF 3.1.7 and BF 3.2.0.

closed 2017-05-26 by richard-scott

richard-scott, 2017-05-30:
Oooh, I've just tried to reproduce the problem so I could document a new issue report and with the latest release 3.2.0 #323d0ae I can't reproduce it. Things are looking up :-)

coreysnyder, 2020-04-17:
@richard-scott I like that you came to say that the command has changed but didn't offer the right one to save everyone else the trouble.

richard-scott, 2020-09-02:
To be fair, its listed in the [Release Notes](https://github.com/betaflight/betaflight/wiki/BetaFlight-3.2)

> **Name changes**
> spektrum_sat_bind_autorst ==> spektrum_sat_bind_autoreset