betaflight/betaflight #3155: BF 3.2.0 - Colibri Race rebooting

Telnoi, 2017-05-26:
FC (colibri race v2) continuously reboots after flashing BF 3.2.

BayaroX, 2017-06-04:
Same issue here at two Colibri Race v2. After flashing 3.2.0 the Colibri is continuously rebooting. No chance to use passthrough for my BLheli32 ESC  :(

Please fix! Thanks in advance!
-Chris-

jflyper, 2017-06-07:
@BayaroX 
> EDIT: Last tested with this pre release >> #1​31​5 from 03.06.2017 03:09

Was this working or not working?

BayaroX, 2017-06-07:
Not working. If it would work, then I would have already joyfully written  :) 

And ... since release #1315 to the current release #1323 there was no change to the code which could have changed something about it.

jflyper, 2017-06-07:
@BayaroX Do you have a reference for the (last, if possible) working version?

BayaroX, 2017-06-07:
Last Stable Version (Betaflight 3.1.7) is working perfect on Colibri Race v2.

I only need Pre-Version of Betaflight 3.2.0 for use of passthrough to my new BLheli32 ESC.

Telnoi, 2017-06-08:
Issue is most likely caused by the implementation of the adaptive notch filter. There is a reference in that feature discussion mentioning that it breaks the colibri race target.

Telnoi, 2017-06-08:
https://github.com/betaflight/betaflight/pull/3074

rav-rav, 2017-06-08:
Colibri was broken because of new box modes, but these were removed.
I can't see any changes that could break this target but I also don't have a colibri to test it.

Can you try a build just before this was merged in?

mikeller, 2017-06-08:
@Telnoi: That issue was keeping the COLIBRI_RACE from compiling, so it was definitely fixed.

BayaroX, 2017-06-08:
Which pre-Release build should i try?    [#1259](http://andwho.sytes.net:8080/job/BorisB_BetaFlight/1259/)  ?!

BayaroX, 2017-06-08:
With build pre-build 1272 the Colibri Race v2 does have the "reboot-loop" problem too.

mikeller, 2017-06-08:
Thanks for testing, @BayaroX. Are you set up to build the firmware yourself?

BayaroX, 2017-06-08:
No, i only downloaded the zip/hex from this build. Was it false? Don`t know how to compile the build myself  :(   Had some linux knowledge, but not tooo much  :)

With pre-build 985 (second release of 3.2.0) the colibri works fine (but no esc passthrough). I now wanted to try to find the last build where the colibri race v2 still works to identify the build which "killed" the colibri...

mikeller, 2017-06-08:
No, it was the right thing to to test with the earliest available version. Unfortunately we do not keep all of the builds, so without being able to build it will be hard for you to test further back. :-(

AndersHoglund, 2017-06-08:
Well, old builds are available on Jenkins. It would be very good if you could do a manual "bi-sect" from build no 985 and fwd. There is always a ref from a build no to a git commit hash, so it will be possible to find the culprit commit. Here are an example for build 985, commit rev 347af510245dac74f91c8eac1afa52adfe949078 , http://andwho.sytes.net:8080/job/BorisB_BetaFlight/985/
I do not have any Colibri Race FC so I cannot be of much more help than supplying the builds.

BayaroX, 2017-06-08:
FOUND  :)

Build 1133 is the last build which is working with colibri race v2.
Build 1134 has the "reboot loop".
Link to build 1134 >> http://andwho.sytes.net:8080/job/BorisB_BetaFlight/1134/

AndersHoglund, 2017-06-08:
Good find. Too bad there are 9 commits in build 1134. I will try and rebuild all 9 one-by-one and post the hex files here, unless you can do a real git bi-sect and re-build yourself.

AndersHoglund, 2017-06-08:
Here are separate builds for the 9 commits in build 1134. Please try them out. 
[betaflight_3.2.0_COLIBRI_RACE_build_1134_parts.zip](https://github.com/betaflight/betaflight/files/1061828/betaflight_3.2.0_COLIBRI_RACE_build_1134_parts.zip)

BayaroX, 2017-06-08:
Thank you very much for your effort! With 2 of the 9 builds, the colibri has the reboot problem. The builds which does **not work** are these two:

**betaflight_3.2.0_COLIBRI_RACE_a45e974c383958af52c390504548d30f84449ea3
betaflight_3.2.0_COLIBRI_RACE_5ab2117a692398267463676029e869a9b3d6a7d3**

With these two builds the colibri race v2 is blinking and beeping **S.O.S.** - Exactly as with the actual/newest build.

BayaroX, 2017-06-08:
@hydra:  Please take a look at these two commits, which "broke" betaflight on **Colibri Race V2** in "**TBS FPVision Elite Bundle**":

https://github.com/betaflight/betaflight/commit/5ab2117a692398267463676029e869a9b3d6a7d3
https://github.com/betaflight/betaflight/commit/a45e974c383958af52c390504548d30f84449ea3

Thanks in advance!

mikeller, 2017-06-08:
Sorry @AndersHoglund, I was not aware that older builds are available. You gotta know that there is this auto-hiding scrollbar in order to actually use it:

[image]

Is there a way to change the theme so this becomes  permanently visible?

Kind of telling how well TBS is supporting their customers. These changes were made after a request to TBS to update their custom code, because it was very outdated. But they did not seem to bother testing the changes, or letting anybody know that they broke support for their boards...

AndersHoglund, 2017-06-08:
@mikeller ,Yeah, old Jenkins builds are a well kept semi hidden secret....
@BayaroX , Have you tried latest CF build? http://andwho.sytes.net:8080/job/Hydra_CleanFlight/
This issue _might_ be a BF - CF mismatch isssue.

mikeller, 2017-06-08:
@BayaroX: Try this, and let me know if it flies: 
[betaflight_3.2.0_COLIBRI_RACE.hex.zip](https://github.com/betaflight/betaflight/files/1062307/betaflight_3.2.0_COLIBRI_RACE.hex.zip)

BayaroX, 2017-06-08:
@mikeller:   This build works fine - no reboot loop!!! Including BLheli32 passthrough!!!  Thank you soooo much!!!
Try to fly this build tomorrow...first it`s time to go to bed for me - sorry (it`s late in germany).

mikeller, 2017-06-08:
Hehe, I know the feeling. Will submit a PR later on today. Thanks for narrowing this down!

borisbstyle, 2017-06-08:
Good job all. What was broken there @mikeller?

AndersHoglund, 2017-06-08:
Good job @mikeller finding the fix and @BayaroX for all the testing effort. And of course @Telnoi for the initial report. Multi site Team Work at its best.

BayaroX, 2017-06-08:
Strange ... after I have made my settings in betaflight, i wanted to flash a new blheli32 firmware and now i could no longer connect to my esc with blheli32. The betaflight configurator still works fine.

I will examine it tomorrow more precisely and then report again. Until then you should perhaps not merge it in the "official" pre-release.

AndersHoglund, 2017-06-08:
No reason to not do a PR if there is a fix for this issue. If there are multiple issues, more PRs to come. Fixing one issue at a time.

BayaroX, 2017-06-08:
I said nothing :) With BLheli32 v32.0.0.1 all is working fine !! Only with BLheli32 v32.0.0.4 i can`t connect to my esc, there must be another problem witch not belongs to colibri :)

So this build works perfectly with Colibri Race v2 and could go public  :)  --  The BLheli32 problem is another problem  :)

mikeller, 2017-06-08:
@borisbstyle: The I2C init was done in the wrong place. Will rewrite it to use generic `targetBusInit()` for this.

@BayaroX: I think there are issues with ESC passthrough in 3.2 at the moment, so this is not specific to COLIBRI_RACE.

mikeller, 2017-06-09:
@BayaroX: Can you try again with this?
[betaflight_3.2.0_COLIBRI_RACE.hex.zip](https://github.com/betaflight/betaflight/files/1063202/betaflight_3.2.0_COLIBRI_RACE.hex.zip)

mikeller, 2017-06-09:
@BayaroX: Can you please try this version?

[betaflight_3.2.0_COLIBRI_RACE.hex.zip](https://github.com/betaflight/betaflight/files/1063224/betaflight_3.2.0_COLIBRI_RACE.hex.zip)

BayaroX, 2017-06-09:
@mikeller: This build doesn`t work! Colibri Race v2 stuck in reboot loop!

Telnoi, 2017-06-10:
@mikeller: the second last build you posted works/blheli_s passthrough also, but vbat or bst is dead.

BayaroX, 2017-06-10:
@mikeller: I tested the second build from 19.07 o'clock again and again i stuck in reboot loop.

@Telnoi:  Do you use FPVision Elite Bundle too? Or with which parts you use Colibri Race v2?

If it helps >> my quad has following components:
Frame = GEP RC LX5
Motors = Emax RS2205S 2300 kV
ESC = Airbot Wraith32 35A
VTX/OSD = TBS FPVision Elite Bundle
FC = Colibri Race v2
Cam = Foxeer Arrow V3
Battery = SLS Quantum 4S 1500mAh 75C
RX = Spektrum Satellit

mikeller, 2017-06-11:
Sorry guys, it looks like my quick 'fix' that I built before running out of the door simply 'fixed' it by not initialising BST and the I2C bus used for BST - which in itself gives a strong indicator that the culprit is in the I2C initialisation for BST.

I simply cannot debug this without even having a board to test on - also, I feel that it should be up to TBS to fix this: This is proprietary hardware, spending time on it will benefit TBS and only TBS, since nobody else is likely to ever use this design.

My hunch is: Simply disable BST on the COLIBRI_RACE, so that users can use it as a 'normal' flight controller without the proprietary TBS extras, and wait for TBS to fix support for their proprietary design.

hydra, 2017-06-11:
yeah, larry from TBS should take a look, they have all the equipment needed to test and debug it.  

Are the I2C pinouts correct in the target.h file?

mikeller, 2017-06-11:
1. I couldn't tell because I don't have a board to ring them out.
2. Their `target.h` defines `I2C2_SDA_PIN`, whereas `I2C2_SDA` is what's actually used. But it looks like BST is using I2C1, so that shouldn't be the cause for this problem.

Praetor353, 2017-06-19:
Does anyone have blhelisuite32 v32.0.0.1? Have colibri v2, flashed mikellers revision linked above, can connect in betaflight configurator now, but blhelisuite32 v32.0.0.4 does not read escs. Want to try v32.0.0.1 but it is no longer on blheli32.com

the-scriptinator, 2017-06-23:
I've just raised a ticket with TBS on this issue as it is not clear to me if they are looking at this issue or not. 
Is "Larry" involved in this bug as yet (not sure of Larry's username)?

larryho5, 2017-06-24:
@the-scriptinator I think this may cause by someone make some updates on the gyro (not sure about this.)
But I have this Boot issue when the Gyro version return not correct back to v1.x or v2.x cleanflight/betaflight firmware.

BayaroX, 2017-07-02:
Is there anything i can do/help to get this problem solved in the next time?
Who can/"should" solve this problem? TBS or Betaflight team? 

If someone from Betaflight team could solve the problem final, if he had a Colibri Race V2, i would be willing to sponsor this person a Colibri Race V2!

I would really like to test more with Betaflight 3.2.0. Also regarding BLheli32 and other features.
I always have new hardware/technique and i am willing to spend time and help, but i am unfortunately forced to sit around and wait. I'm not so good at sitting around and waiting  :-D

Kind regards - Chris

mikeller, 2017-07-02:
@BayaroX: Probably the best think you can do is to raise yet another support case with TBS, to show them that it matters to their customers if support for their proprietary hardware is broken in Betaflight.

It is very kind of you that you offer to sponsor hardware, but in this case I think this is not a solution. BST and the hardware it runs on was designed by TBS, and will only ever be supported by their hardware. If the developers who volunteer their time to work on Betaflight have to chose between using their time to implement / fix stuff that will help all users, and using it to fix stuff that will only help TBS customers, I think for most of us, the former will take precedence - and there is always something to do.

In this respect, TBS has to realise that if they want to add proprietary hardware, the buck stops with them to make sure their customers have support for that proprietary hardware. The 'we'll throw it onto the market; and let Betaflight support it' that they (and a lot of other manufacturers) show just does not cut it.

the-scriptinator, 2017-07-03:
I have an email dated Jun 24 from Trappy's email account signed by "Remo" which says 

"Thank you for the hint. The issue already got resolved. Seems like you flying latest master branch. Latest release is 3.1.7 which works fine on colibri race. We recommend to stick to official releases and not using developers daily builds. 
Best regards,
Remo"

@mikeller I assume that as far as the Betaflight project is concerned this bug is still open and still TBS's responsibility? (as far as I can tell you're a committer - I assume you're able to speak for the BF project??).

mikeller, 2017-07-03:
I can't test, since I don't have the hardware, but if it does not work in current master then it is a bug as far as Betaflight is concerned, yes. As for their custom extension being TBS's thing, and not ours: Betaflight only implements one side of this protocol, so we don't have any stake in keeping it alive, and no good way of testing if it is fit for purpose.

I really think that 'it works in the latest release, so we are not concerned about it' is a lousy attitude on TBS's side. This means that nobody can test and provide feedback for the next release, and depending on when we are starting to release release candidates, support for this board will probably be broken in them, and potential other issues not fixed, since it is untestable.

If they really want to support Betaflight on their hardware, I suggest that they look into fixing support for it in master as soon as it breaks, not just at the last moment to make it into a release. :-(

BayaroX, 2017-07-03:
I can understand mike there already well. TBS does not have its own software and is based on Betaflight, which I can also flash over the TBS Agent. But why should Betaflight take care that the colibri continues to work when TBS is actually responsible? TBS wants your customers to use the TBS flight controllers with Betaflight, so please also TBS should do something for it and do not rely on Betaflight to do the work for TBS!

I do not know exactly where the problem is. But if TBS incorporates something "special", then TBS must also ensure that it works with Betaflight! And not only with old Betaflight versions, but also with new versions! I as TBS customer would like to use no only old Betaflight versions, but of course current software with current features etc ...

I hope you understand what I mean and hope TBS does his work. Otherwise, I will not be the only customer who will not use a TBS product in the future.

Betaflight is currently the best software for flightcontrollers. If TBS wants to be there, TBS should also do something for it!

Just my 2 cents...
-Chris-

blckmn, 2017-07-03:
I have a COLIBRI, and will be looking into the issue on behalf of TBS shortly. It is just a matter of time - hopefully this weekend.

the-scriptinator, 2017-07-04:
@BayaroX  - I think mikeller had it right when he suggested you also open a ticket with betaflight.
The approach from the reply I got seemed to be "we know best you jsut run what we tell you" ignoring the fact that the world has moved on and 3.1.7  does not work well with the latest production BLheli.

I agree with mikeller - if they are not looking at the bugs now, then we can expect our hardware to be non-botoable when 3.2 releases.

I've specifically asked for an assurance that BF3.2 will run on the Collibri on day one of the release and hope that other Collibri users will raise a ticket and ask TBS the same question.

BayaroX, 2017-07-20:
@blckmn - Any news for us? Still working on this problem? Could you find/fix something?

blckmn, 2017-07-22:
https://github.com/betaflight/betaflight/pull/3598

@BayaroX can you please pull that request and see if it is BST causing the issue. I am travelling at the moment, and don't have a COLIBRI to test with. If its BST then I will look into where the problem is with that code. Thanks :)

BayaroX, 2017-07-23:
@blckmn - I cant compile a BetaFlight release myself. Im willing to test, but i need a compiled build (hex or something else) for download and flashing it to my colibri v2.

AndersHoglund, 2017-07-23:
Pulled and build it for you:
[betaflight_3.2.0_COLIBRI_RACE- a4de1df.zip](https://github.com/betaflight/betaflight/files/1168153/betaflight_3.2.0_COLIBRI_RACE-.a4de1df.zip)

For reference, sources on this branch: https://github.com/AndersHoglund/betaflight/tree/bst_disable

BayaroX, 2017-07-23:
@AndersHoglund - Thanks a lot! Downloaded this build, flashed it - it is working fine!   :)
@blckmn - Looks like BST causes the reboot-loop btw the problem.

the-scriptinator, 2017-08-14:
@larryho5 -hey Larry any movement on this issue  or maybe an ETA?

blckmn, 2017-08-14:
@the-scriptinator the existing 3.2 RC works fine for Colibri Race but without BST.

I've raised a PR https://github.com/betaflight/betaflight/pull/3816 for fixing this up, but I don't have any BST setup, so am unable to test. If you could test it, that'd be great.

blckmn, 2017-08-14:
Actually turns out I have a power cube bundle (kindly supplied by TBS) so I'll be testing in the coming days.

the-scriptinator, 2017-08-15:
@blckmn - thanks for the update.

I have a powercube bundle and would be happy to test if that is useful and not a pain for you.

BayaroX, 2017-08-21:
Actual BF 3.2.0 RC3 version is working almost fine with Colibri Race v2 in TBS FPVision Elite Bundle.....all is working fine (arm, flying around, ...), expect of sending transmitter stick inputs from Colibri Race v2 to TBS VTX >> this means >> no chance to enter OSD menu, exit Pit Mode or do first start calibration of TBS FPVision with transmitter-sticks.

I resetted my TBS FPVision and after reboot it said >> (translated in own words) >> can`t calibrate, cause no transmitter signal. So i had to flash BF 3.1.7, calibrate FPVision and flash BF 3.2.0 RC3 again.

Is this because of actual non working BST? Hope there will be a final solution soon. Pit Mode and OSD menu are very cool features in TBS FPVision.

Telnoi, 2017-08-21:
Short answer, yes it is because of disable BST

[11 more replies]

closed 2017-08-24 by martinbudden