betaflight/betaflight #3160: FlyingLemon Kiwi F4 v2 and Plum F4 unresolved issues

peteoz, 2017-05-27:
before I start just want to mention that I got 5x Kiwi F4 v2 and 2x Plum F4 of which 4 Kiwis and 1 Plum are installed (3 in my quads and 2 in my friends' quads) so these issues are not specific to just one single build.
Also, I've already read the following threads and there are no solutions to issues that I'm going to describe below :
https://github.com/betaflight/betaflight/issues/3082
https://github.com/betaflight/betaflight/pull/2858

(1) Kiwi F4 v2 : 
tested 4 different firmware versions, each comes with its own issues, some issues fixed in some versions but new issues introduced.
(a) latest official (stable) release 3.1.7
- SD card reader no longer recognized
- LEDs not working (LEDs were supposedly fixed in subsequent 3.1.7 version available for download from FlyingLemon website, but new issues introduced which makes this version unusable, see (b) below)
(b) latest version available for download from FlyingLemon website, also 3.1.7 but this is later release from May.
- LEDs apparently fixed by having to use VTX.DTA for LED signal wire
- SBUS no longer working with Frsky XSR Rx (tried both sbus_inversion ON and OFF), however Telemetry is working fine
(c) another version 3.1.7 dated April 10 and sent to me by email directly from FlyingLemon
- LEDs working OK but motors 1 and 4 stopped working
- Frsky XSR Rx SBUS not working (telemetry is OK)
(d) original version 3.2 as distributed on the new board
- LEDs not working, everything else is OK (SDcard reader fine, SBUS and telemetry on FrSky XSR Rx are fine)

(b) & (c) above issue affecting FrSky XSR Rx SBUS, this issue does not exist in FW version (a) & (d), also where FrSky XSR SBUS fails, iBUS (tested with FlySky Rx) is actually working

based on the info above, the best available FW at this point in time is still the original 3.2 as distributed with the board because only LEDs are not working but everything else is OK (XSR Rx and SDcard reader)

(Plum F4)
(a) latest official (stable) release 3.1.7
- LEDs not working and I cannot figure out any work around for this, this board has no option for VTX.DTA, on my build I only have 2 pads available to try to remap : RSSI monitor and Current ADC pad, I tried to reallocate resources but then the board freezes and the only way to fix this, is to force bootloader and reflash&erase the board with new FW

(b) original version 3.2 as distributed with the board
- FrSky XSR Rx SBUS not working (telemetry is OK)

I contacted FlyingLemon week ago but got no response.

Thanks to anyone looking into these issues.
Cheers
Pete

mikeller, 2017-05-28:
Hmm, normally @flyinglemonfpv is quite responsive to support requests.

I know there have been a number of updates to these board definitions, so chances are that the latest versions of 3.2 are working, you could try the ones published here: http://andwho.sytes.net:8080/job/BorisB_BetaFlight/lastSuccessfulBuild/artifact/obj/

flyinglemonfpv, 2017-05-28:
Problems with sbus, SDcard and leds are resolved in my build of 3.1.7, available on product page. Led must be connected to VTX.DTA

martinbudden, 2017-05-28:
@flyinglemonfpv , it would be good if you could make a PR so the issues could be resolved in betaflight itself, so the fixes are in 3.2 when it comes out.

flyinglemonfpv, 2017-05-28:
I made it. All changes are merged.

flyinglemonfpv, 2017-05-28:
I made it. All changes merged.

peteoz, 2017-05-28:
@flyinglemonfpv
> Problems with sbus, SDcard and leds are resolved in my build of 3.1.7, available on product page. Led must be connected to VTX.DTA

just to provide some feedback : 

the newly updated version 3.1.7 (12c4211) available from product page - tested on one of my quads.
- Frsky XSR Rx SBUS (& Telemetry) working OK
- SD card reader working OK
- at the moment I'm unable to test LEDs because my signal wire was connected to LED pad and to access VTX.DTA pad on this board is not easy because it's on the reversed side and not accessible from side either once installed.

can you confirm that from now on all the future FW releases will require LED signal wire to connect to VTX.DTA and not back to LED pad at some point in the future (ie. BF 3.2) ?

secondly, 
is there any work around to get the LEDs to work on Plum.F4 FC (there is no "VTX.DTA" pad) ?
on my build I still have "RSSI" and "Current ADC" pads available, can I remap resources and get one of these pads to act as LED pad ? (I tried but can't get it to work, the board freezes wehn I remap resources)

thanks
Pete

flyinglemonfpv, 2017-05-29:
With FW 3.2 led will be on LED pad, but if you want on VTX.DTA just let me know.

peteoz, 2017-05-29:
> @flyinglemonfpv : With FW 3.2 led will be on LED pad, but if you want on VTX.DTA just let me know.

the problem is with all the people who bought this FC and were asked to rewire the LED signal wire and after 3.2 is released be asked again to rewire to the original pad ? I don't know, I bought 5 of these FCs and I was rewiring 3 of my quads today. both LED pad and VTX.DTA pad are difficult to access when the FC is mounted in standard way then these pads are at the back where most of the other wiring is, also right below these pads are my battery connections on PDB or 4-1 ESCs with a capacitor. I had to unsolder many wires to make this mod and then resolder again.
It sounds easy to ask to rewire 1 signal wire, but on this FC it's actually quite a messy job.

I would be quite happy after 3.2 release (and if you move the LED functionality back to LED pad) that you allow those who already modified to VTX.DTA to be at least able to remap resource and continue to use VTX.DTA

I also noted that with this newly fixed release that somehow there is a massive improvement on my gyro traces, didn't think it was possible to address this with a FW upgrade, I always thought this was hardware related. In the previous 4 firmware versions that I tried my blackbox readout was showing insane amount of noise and spikes mainly on the yaw gyro traces but also on pitch and roll, after this release my blackbox readouts are super clean just like was always the case with Kiwi version 1. I don't know what you changed but whatever it is please keep it that way for 3.2 release.
(this again was not related to one specific build, this fix helped on all of my quads with the same FC so really the noise was related to previous FW versions)

johnmoto1, 2017-06-05:
OSD not working , go into betaflight osd hit the upload button it says uploading and stays hanging there does nothing have no osd on my screen , have the smart audio working , I am fairly shaw I have setup everything corect in the ports tab but can't be sure as there is no information on how to do this anywhere icluding your website were it should be , I await your reply.
Regards John

mikeller, 2017-06-05:
Hi @johnmoto1.

Please see my comments to your post on #2858.

Also, please avoid cross-posting the same content into different issues. Before posting, take the time to find the issue is most appropriate for your comment, and then post only there.

kaarleenzz, 2017-06-15:
I'm also having horrible vibration issues. I'm using the version provided on the flyinglemon website.

What is the 'best' firmware to use at the moment? The one I had on by default was not working with sbus.

mikeller, 2017-06-15:
@kaarleenzz: See @flyinglemonfpv's post above. There are latest versions of the firmware available on the product page.

johnmoto1, 2017-06-16:
Hi mate the only firmware is on flying lemons site if you are having vibration problems you must soft mount everything motors, fc , pdb

[image1.JPG]
[image2.JPG]
From John Daniels

kaarleenzz, 2017-06-16:
I have soft-mounted the PDB and flight controller. This one has an mpu6000 gyro so it shouldnt behave this horribly, I'm thinking that there is something more going on. 

I basically replaced a FC (which had 6500 gyro) with this one and this is unflyable, while the other had minor issues at full throttle sometimes. 

I also can't log anything in the blackbox because none of the FW version I've tried seem to work with SD card...

At this point I'm kind of regretting switching over from my 15$ FC...

Will try to fly with betaflight 3.2. Blackbox and beeper doesnt work on this version, but if the flight is fixed, I don't care.

peteoz, 2017-06-16:
@kaarleenzz , I'm wondering if you are running latest FW version. I know you mentioned you downloaded it from FlyingLemon website, but when did you download it ? The previous version that was there had the SD card issue that you are still having, but this was fixed in the new version.

can you run "dump" or "diff" command in CLI to confirm you have the correct version, it should give you this 
"# version"
"# Betaflight / KIWIF4V2 3.1.7 May 25 2017 / 09:14:50 (12c4211)"

All the issues that I mentioned in the original post of this thread were addressed in this FW release. In one of my posts above I mentioned that previous versions of the FW were showing noise on gyro blackbox readout (mainly on the yaw axis), this was also fixed in latest FW (but having said that, the noise was filtered out by LPF anyway and was not causing any vibrations on any of my quads even before the fix).

Maybe just to summarize and simplify this thread in regards to what was reported in the original post
(1) KIWI.F4 v2
- SBUS on the Frsky XSR - fixed
- SD card reader - fixed
- LEDs - fixed (when running off the VTX.DTA pad)
- in addition the Gyro noise (that was not reported in the original post) was also fixed
(tested on 3 different KIWI.F4 boards in 3 different quads)

outstanding : would like to hear confirmation what pad on the board will be used for LEDs in future FW versions (3.2 and up), and if it is reverted back to LED pad, whether the users who already wired LED signal wire to VTX.DTA can continue to use this pad ie. with resource remapping

(2) PLUM F4
- LEDs not working - not fixed
outstanding - confirmation whether or not this is fixable in future FW versions so that users know whether to install addressable LEDs (in case this is not fixable users would make a decision to install non-addressable LEDs instead)

kaarleenzz, 2017-06-16:
Yes, mine gives: 
# version
# Betaflight / KIWIF4V2 3.1.7 May 25 2017 / 09:14:50 (12c4211)

So I'm guessing it is the correct one.

The blackbox tab just shows:
[image]

Configuration tab:
[image]

I'm trying to use a 64gb sd card. Could that be the problem? Does it have to be formatted in any particular way? I will probably try to hard mount everything next and see if that helps...

mikeller, 2017-06-16:
Never tried a 64GB card. Do you have a smaller one that you could use to try if that makes a difference?

Also, in general, for streaming any kind of data, best performance can be achieved by using the official SD card formatting utility: https://www.sdcard.org/downloads/formatter_4/

peteoz, 2017-06-16:
same here, I don't use large SD cards. I use small cheap 2GB cards, in fact I have this one in all of my quads now : 
https://www.amazon.com/gp/product/B000VOU91U/

I remember once trying 64GB on a different board and was having some issues with it, besides the SD card readers on FCs (as far as I now) use only 4GB (or 2GB) and the rest is wasted.

all my 2GB SD cards I formatted as "MS DOS Fat" on my Mac.

kaarleenzz, 2017-06-16:
Will try to get a smaller SD card and see if it makes a difference.

johnmoto1, 2017-06-20:
I was having same issues as you it wasn't until I learned of the 3.2 fw on flying lemons website 
That everything worked unless you download flying lemons fw it's not going to work.

BlackPixxel, 2017-06-22:
It is not exactly Betaflight-related, but here is the only place where I found other pilots with this board.

Does your Kiwi V2 also get so damn hot on the F4 CPU when the board is powered by the Lipo for a few minutes? I almost cannot touch the CPU then.

maverikou, 2017-08-05:
Latest BF version works fine on KIWI.F4 v2. What remaining problems are you having?

jflyper, 2017-08-05:
@BlackPixxel Open a kiwi/plum thread in RCG ;) Please be reminded that the Betaflight github issues is the place for reporting and discussion that make the Betaflight software better.

peteoz, 2017-09-15:
Hi,

I have finally went ahead and installed BF 3.2 (RC5) on my quads, the LEDs are no longer working again on Kiwi.F4.

-------------------------------------------
@flyinglemonfpv : With FW 3.2 led will be on LED pad, but if you want on VTX.DTA just let me know.
-------------------------------------------

The manufacturer left this out with no definite answer as to what will happen when BF3.2 is released.
So I'm asking again, can we get a closure on this issue and final confirmation on how to get LEDs working on BF3.2 ?

and btw. the same question in regards to Plum F4 was never answered.

Thanks

mikeller, 2017-09-16:
@peteoz: @flyinglemonfpv would probably know the answer to this.

peteoz, 2017-10-04:
@mikeller : I believe @flyinglemonfpv is subscribed to this thread since he was previously responding here, but last time he was seen here was back in May. since then I tried to contact him directly by email also regarding other matters but can get any response.

in the meantime I re-wired some of my quads again and can't get the LED to work on BF3.2 (tried RC4, RC5 and today RC6).

since we received no definitive confirmation from @flyinglemonfpv whether LED should be wired back to LED pad with B3.2, or left on VTX.DTA pad as that's where it was working on BF3.17, I went ahead and tried both options, LED strip is currently not working on either of these two pads.

in BF3.2 the resource assigned to LED strip is A00, whereas in BF3.17 it was B07 (when running of the VTX.DTA pad), I tried to remap LED back to B07 as that's where it was by default in 3.17 but then my board freezes and I can only get in by forcing bootloader and reflashing the board.

anyway, what can be done for this to get back to working order as it was in BF3.17 in the absence of @flyinglemonfpv ?
all we know it was working just fine (using VTX.DTA pad) prior to release of BF3.2, whatever change was done between the two BF versions, is what's causing the problem. can we get it back to a state where it was previously working ?

thanks

mikeller, 2017-10-04:
Not sure how much significance this has, but can you try changing `TIMER_OUTPUT_NONE` to `TIMER_OUTPUT_STANDARD` in https://github.com/betaflight/betaflight/blob/master/src/main/target/KIWIF4/target.c#L35 ?

peteoz, 2017-10-04:
@mikeller 
I suppose that's not something I could do through CLI ?
this is the source code that needs to be compiled after modification to make a new HEX file ?
I'm just not familiar with the process, is there any tutorial on how to get me started ?

mikeller, 2017-10-04:
@peteoz: Ah, I assumed you were set up to build the firmware. Will look into it tonight and post a binary to test.

peteoz, 2017-10-04:
thank you @mikeller

jflyper, 2017-10-04:
@peteoz 
Which board do you have the LED issue with? KIWIF4, KIWIF4V2 or PLUMF4?

peteoz, 2017-10-04:
@jflyper I have all of these boards (2x KIWIF4, 3x KIWIF4V2, 2x PLUMF4), but the issue I'm referring to in my last post is regarding KIWIF4V2. 

1)
the older original KIWIF4 I cannot test with BF3.2, I installed those almost year ago and both quads are not in working order at the moment, but there weren't never any issues with the original KIWIF4 up to version BF3.17 which was the last BF version I was running, everything always worked on those, incl. LEDs

2) 
KIWIF4 V2 - everything I reported in the original (first) post of this thread was eventually fixed but there was never an official version released with the fixes. The version with all the fixes is this one "Betaflight / KIWIF4V2 3.1.7 May 25 2017 / 09:14:50 (12c4211)" and it's only available for download from KIWIF4 V2 product page (flyinglemon.eu website), somehow this version never made it to official BF release.
the LED strip signal wire had to be wired to "VTX.DTA" pad for it to work.
in regards to the issue with XSR Sbus not working in some versions, this was also fixed but it is something I was unable to test in BF3.2 because since then I have upgraded all my Kiwi.F4v2 quads to Crossfire Rx, I'm happy to put XSR back in to one of my quads to verify if it is still working in BF3.2.
btw. in the process of upgrading the quads from Frsky Rx to Crossfire Rx I found that there is only 1 fully functional UART on this board where both Rx and Tx are working, that's UART3.
UART1 Rx (Sbus) is inverted, and the last UART6 which has both Rx and Tx pads is actually not working when I tried to connect Crossfire to it (I need both Rx and Tx for Crossfire). It's not a big deal, I used UART3 for my Crossfire, but it's just weird that UART6 is not working, must be hardware issue, maybe that UART is also inverted, I don't know.  

3)
PLUM F4 - as reported in the original post, the LED strip is not working on this board, there was never fix nor suggestion how to deal with that. I'm now running 5V non-addressable LEDs on my quads with PLUM F4 FC.

jflyper, 2017-10-04:
**_Kiwi F4 V2_**
- From the description and wiring diagram, UART6 has a circuit branching out for SmartPort connetion, and  I suspect this branch circuit is interfering with full full-duplex operation on UART6. AFAIK, it is impossible to make the branch circuit co-exist with normal full-duplex UART. So yes, UART3 is the only fully functional full-duplex UART you have on this board.

Kiwi F4V V2 and Plum F4
- Now, the LED strip. These two boards have consistently defined PA0 as LED strip signal pin. I think this is connected to LED pad. I see Kiwi F4 V2 has a LED pad. It doesn't work?

mikeller, 2017-10-04:
@peteoz: I had a quick check with the KIWIF4V2, and the LED pad does not seem to work with the vanilla firmware indeed. I also couldn't figure out how to make it work, but I managed to get LED_STRIP to work on VTX.DTA (which does not seem to be used for anything else at the moment anyway) with this custom firmware:

[betaflight_3.2.0_KIWIF4V2.zip](https://github.com/betaflight/betaflight/files/1356127/betaflight_3.2.0_KIWIF4V2.zip)

Please let me know if this is working for you, and I can change this for 3.2.

peteoz, 2017-10-04:
@jflyper , UART6 not working is not a big deal, I hooked up my Crossfire Rx to UART3 and was hoping to connect my BLheli32 ESCs telemetry to UART6, but ESC telemetry is still unknown territory for me, not much info around, I'll wait for that and do it on my next build.

it's unfortunate that so many great F4 boards with MPU6000 were "crippled" to make them compatible with Frsky receivers. it's impossible to find one with 3 (and sometimes even with 2) fully functional UARTs.

and yes, that's correct, none of the FW versions defined as PA0 for LED strip ever worked, the only one version that worked had PB7 instead and the LED strip was required to be wired to "VTX.DTA" pad rather than LED pad.

@mikeller just posted custom FW implemented just like that, I'm just about to go to test it and will let you know.

jflyper, 2017-10-04:
I'm sooooo seriously and curiously and eager to know where the trace from LED pad is going!!! 😉 

Same mistakes three times in a row is somewhat unimaginable.

peteoz, 2017-10-04:
@mikeller thank you for the custom FW, it works.

tested on 2 quads.
(1)
- LEDs work when VTX.DTA pad is used, slightly different behaviour than on 3.17 but I don't suspect this is due to the custom FW, more likely there is a difference in 3.2 the way Color Modifier works with Throttle (previously going to full throttle the LEDs would go through the entire color spectrum, whereas now it goes from base colour to green colour at full throttle),
also Overlay/Indicator not working when turning (previously assigned LEDs would blink when turning right or left). like I said this may be just a different behaviour in BF 3.2 and it doesn't concern me.

(2)
SD card working fine and getting BB logs - only tested on bench, will report back if anything different once tested in-flight

3)
Crossfire Micro v2 Rx working fine

4)
I installed back XSR receiver on one of the quads to test the functionality since previously Sbus was not working in some earlier versions. It works all fine including telemetry but user is required to set "tlm_halfduplex = OFF" in CLI (default is "ON") for telemetry to work.

mikeller, 2017-10-05:
@peteoz: Thanks for testing. #4285 moves the LED_STRIP output back to VTX.DTA.

PR #4285 (2017-10-09): Moved LED_STRIP pin back to VTX.DTA for KIWIF4V2 / PLUMF4.
Fixes #3160.
closed 2017-10-09 by mikeller