betaflight/betaflight #3167: Accidentally Enabled "No Reboot Sequence" in DFU Mode

rinozen, 2017-05-28:
please help me... im trying to flash my F4 in DFU Mode, but i noticed that "no reboot sequence" and "full chip erase" is enabled... now my FC cant be detected in device manager.... please help me how to fix it???

jflyper, 2017-05-28:
This is a place report software issues. Please use the board specific thread in RCG (or elsewhere) for support request. Thank you.

rinozen, 2017-05-28:
please give me the link

jflyper, 2017-05-28:
@rinozen What is your FC?

rinozen, 2017-05-28:
F4 Revo

jflyper, 2017-05-28:
??? Product URL?

AndersHoglund, 2017-05-28:
"Full Chip Erase" does not work with F4's. Will hang during flashing. A known problem.

You need to re-flash by forcing the FC into bootloader mode by having boot pads bridged when connecting USB. Select  "No reboot Seq" (as you already are in boot loader mode) select the REVO firmware and version. Hit "Load Firmware online" and then "Flash Firmware". Do not forget to removed the boot pad bridge.

Read all about flashing in the Wiki: https://github.com/betaflight/betaflight/wiki/Installing-Betaflight

RCG is another source for info and the place for support: https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread/page3243
As well as FB: https://www.facebook.com/groups/betaflight/

rinozen, 2017-05-28:
jflyper : https://www.aliexpress.com/item/F4-Flight-Controller-SP-Racing-F4-Flip32-FC-Opensource-Betaflight-for-FPV-Racing-Drone-Quadcopter-Multirotor/32801771929.html
anders : i dont know wheres the boot pin coz it have only tacticle boot button

jflyper, 2017-05-28:
@rinozen I believe it's AIRBOT F4.

https://www.rcgroups.com/forums/showthread.php?2731917-Airbot-F4-Flip32-F4-Thread

AndersHoglund, 2017-05-28:
Well, then it is simply a matter of keeping the button pressed while connecting.

robert-hoffmann, 2017-08-16:
"Full Chip Erase" does not work with F4's. Will hang during flashing. A known problem.

Didn't pose a problem on my matek f405 aio ..got lucky i suppose!

AndersHoglund, 2017-08-16:
The "Full Chip Erase" problem was fixed some time ago.

closed 2017-08-16 by AndersHoglund