betaflight/betaflight #3179: vbat data wrong with DYS F4 pro FC

scottinchina, 2017-05-29:
Hi all, I update a new FC with DYS F4 pro, the vbat get wrong data when I connected with Lipo. It should be at 16.x volt with fully charged 4S battery, but only 5v show up (same in OSD). By tunning votage scale, the scale range between 0-255, still can not up to 16.x volt. 
I also test with Foxeer HS1190 which can detect votage directly, the votage is correct. 
Anyone know how to solve this issue? 
Thanks!~~

Here is the firmware detail:
Flight controller info, identifier: BTFL, version: 3.1.7
2017-05-29 @ 23:16:07 -- Running firmware released on: May 16 2017 14:05:58
2017-05-29 @ 23:16:07 -- Board: OBF4, version: 0

scottinchina, 2017-05-29:
[image]

scottinchina, 2017-05-29:
plus , did anyone knows the value of current scale and offset ? I don`t have power meter to run the test.

jflyper, 2017-05-30:
I think this is a support request.

Please direct support requests to the appropriate thread in RCG.

https://www.rcgroups.com/forums/showthread.php?2856541-DYS-F4-OSD-BEC-AiO-Flight-controller

scottinchina, 2017-05-30:
ok, already feedback to the manufacturer , thanks anyway

mikeller, 2017-05-30:
@scottinchina: Where do you get the Vbat that you feed into the board from? Is it possible that your PDB has got a Vbat out that is already scaled down by a voltage divider, and there is another voltage divider on the Vbat in pin on your flight controller?

scottinchina, 2017-05-30:
[image]
Here is the FC layout, this board intergrated with PDB also Lipo connect direct to the board, so there is no need to solder any wire, the OSD from Vout same wrong volt scale.  that is really confused me.

closed 2017-05-30 by scottinchina

nkkr77, 2018-01-11:
hey guys
by the way how to make voltage value to display,i also use dys f4 butvi dont know where should i solder.
Please respond as soon as possible.
thanks

mikeller, 2018-01-11:
@nkkr77: GitHub is for questions and discussion around the **development** of Betaflight / cleanflight. For questions around the configuration and use of Betaflight / cleanflight, please visit [RCGroups](https://www.rcgroups.com/forums/search.php?query=betaflight%20cleanflight%20configuration%20use).