betaflight/betaflight #3180: Soft Serial not working on OMNIBUSF4SD target, not defined in target.h

TwoToneEddy, 2017-05-29:
Hi,
I've been trying to get smartport working using soft serial on this board. I noticed in target.h there are no allocations for the pins on this board. I belled out the soft serial pads on the bottom of the board (ch5 and ch6) and they go to PB3 and PB4 respectively. I don't have enough knowledge about the timer allocations, is this something someone could take a look at? I've managed to get smartport working using inav on this board using soft serial. It required ch5 and ch6 to be bridged with a 1K resistor and setting the telemetry inversion to ON.
Thanks
Lee

jflyper, 2017-05-29:
Default pins were deliberately left undefined in target.h; that's what (almost) everything is going to be in the future BF.

But for now, if you are hooking up SmartPort to CH5 or CH6, use one of the following CLI commands.

```
resource serial_tx 11 c8 # for SOFTSERIAL1 on CH5
resource serial_tx 11 c9 # for SOFTSERIAL1 on CH6
resource serial_tx 12 c8 # for SOFTSERIAL2 on CH5
resource serial_tx 12 c9 # for SOFTSERIAL2 on CH6
```

You only need to connect to a single pad of your choice.

P.S. I don't know where you got these from, but PB3 and PB4 are wrong.

TwoToneEddy, 2017-05-29:
Thanks for the clarification. I came to the conclusion that PB3 and PB4 were the correct pins as I found what pins the ch5 and ch6 pads are attached to on the bottom of the board. Looking at the pinout for that processor I found it was PB3 and PB4. I'm not sure how I got that wrong. When you say ch5 and ch6 you're not referring to the motor outputs are you (please don't take offense to that question!). The pads I'm referring to are hidden on the underside of the board.
Thanks again

jflyper, 2017-05-29:
https://github.com/betaflight/betaflight/blob/master/src/main/target/OMNIBUSF4/target.c#L40-L41
```
    DEF_TIM(TIM8,  CH3, PC8,  TIM_USE_PWM,                 TIMER_OUTPUT_NONE,     0), // S5_IN
    DEF_TIM(TIM8,  CH4, PC9,  TIM_USE_PWM,                 TIMER_OUTPUT_NONE,     0), // S6_IN
```

jflyper, 2017-06-01:
@TwoToneEddy You may want to take a look at #3190 also.

closed 2017-10-23 by jflyper