betaflight/betaflight #3181: Unable to setup RSSI range

MathijsG, 2017-05-29:
Hi,

I'm using a FlySky i6 Transmitter with custom firmware which allows me to put the RSSI on an AUX channel.

However the scale of this RSSI is:

0% error = 2000
10% error  = 1900
20% = 1800

....

100% error = 1000

However when mapped the RSSI to this channel I always get 50% error rate read out. Where can I set the the scale and range of this in Betaflight? Having it to be from 2000 to 1000.

mikeller, 2017-05-29:
When looking at the bars on the Receiver tab in the Betaflight configurator, does the bar where the RSSI display is supposed to be actually move?

jflyper, 2017-05-30:
I suspect the RSSI channel is configured incorrectly and picking up a value from a channel that is always neutral. In configurator, it is numbered from 1, and AUX1 is 5 (+4). Range inversion can be set with cli command `set rssi_invert = on`.

MathijsG, 2017-05-30:
I got it working yesterday, just after writing this issue. Partly my lack of knowledge on the subject, and some confusion in the channels indeed.

And because I'm not the only one I want to suggest some clarification. Here a screenshot of how it looks currently:
[image]

Maybe we can make the RSSI channel list dropdown somewhat more clear, maybe also output their current function, the same as in the left list. Now people easily mistake AUX 5 with channel 5, while you need to add the 4 control channels.

jflyper, 2017-05-30:
@MathijsG Can you submit a change request in BFC repo also?

https://github.com/betaflight/betaflight-configurator/issues

MathijsG, 2017-05-30:
Ah I'm sorry, I will try to move this and close it here.

closed 2017-05-30 by MathijsG