betaflight/betaflight #3183: http://www.betaflight.tk refresh to http://185.87.186.121:3000 dead

maglub, 2017-05-30:
Dear,

The redirect on the www.betaflight.tk to sign up to the Slack channel is dead.

```
malu@kmg-mcp001.local:/Users/malu/dev/conextrade/vagrant-kitchen-epr/shared/ansible-epr-inf $curl http://www.betaflight.tk/
<meta http-equiv="refresh" content="0; url=http://185.87.186.121:3000/" />
<!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Transitional//EN" "http://www.w3.org/TR/xhtml1/DTD/xhtml1-transitional.dtd">
<html xmlns="http://www.w3.org/1999/xhtml">
  <head>
  <body>
  <p></p>
  </body>
</html>
```

Following the refresh:

```
malu@kmg-mcp001.local:/Users/malu/dev/conextrade/vagrant-kitchen-epr/shared/ansible-epr-inf $curl http://185.87.186.121:3000
curl: (7) Failed to connect to 185.87.186.121 port 3000: Connection refused
```

Trussroads, 2017-06-10:
There is no one to get code support from on FB either? How do I contact a dev?

bilson, 2017-06-10:
Try with https://betaflightgroup.slack.com/

mikeller, 2017-06-12:
@maglub: http://betaflight.tk/ is working again.

closed 2017-06-12 by mikeller