betaflight/betaflight #3184: BF 3.2.0 - LED 'Warning' beeper mode

TheAngularity, 2017-05-30:
@mikeller pls, can you re-work #3138
the idea is already good ... but the orange warning led came also with the vbat warning ... so vbat warning blinks now orange and red ... i think it was not so wanted ... imho it should only flash orange if the beeper was explicitly activated, so as an additional search function when the quad was lost

mikeller, 2017-05-30:
Yeah, I guess doubling up there isn't the world's best UI design. The user in the feature request explicitly asked for an indication for things like mode changes, so I think having it on only in beeper mode is not what's wanted. My suggestion is to enable it when the beeper is going _except_ when it is for a condition that already has an LED indication.

TheAngularity, 2017-05-30:
> My suggestion is to enable it when the beeper is going except when it is for a condition that already has an LED indication.

yes ... so i had imagined that

TheAngularity, 2017-06-02:
@mikeller tested your #3195 ... this pr fixed now this issue

closed 2017-06-02 by TheAngularity