betaflight/betaflight #3188: Card Emax femto F3 motor 4 not spinning

alexandreBoussemart, 2017-05-30:
Hi everybody,
I am trying to mount a 130. I have mounted an Emax F3 Femto (https://www.banggood.com/fr/Emax-STM32F303-F3-Femto-Flight-Controller-with-Integrated-BECBuzzer-PadsVBatPDB-p-1101875.html) card with betaflight SPRACINGF3EVO 3.1.7. I left all the original configurations and I have my engine 4 that do not want to run. In esc, I have opto pro 20a, in oneshot 125 and in the tab motors all the engines run except the 4. Someone has an idea?
I have already tried with other esc / engine and always the same.
Thank you
Alexandre

jflyper, 2017-05-31:
Hi @alexandreBoussemart 
GitHub is for questions and discussions around the development of the Betaflight firmware. For help with your setup, please search the RCGroups first.
https://www.rcgroups.com/forums/search.php?searchid=61185873&query=emax%20femto
You may also want to search SPRACING F3 EVO thread.

alexandreBoussemart, 2017-06-02:
The material is ok, the card is new. The problem must come from betaflight yet I take the latest version. You have no idea?
I saw that it was necessary to remap when use the DSHOT 600 but I am in ONE SHOT 125.
I can not find any post with my problem on rcgroups.
cordially

jflyper, 2017-06-03:
@alexandreBoussemart 
This is the RCG thread. Please post your support request here.
[SPRacingF3EVO flight controller - CHEAP! F3/SDCard Socket/Race Transponder](https://www.rcgroups.com/forums/showthread.php?2641205-SPRacingF3EVO-flight-controller-CHEAP%21-F3-SDCard-Socket-Race-Transponder)

closed 2017-10-22 by jflyper