betaflight/betaflight #3191: 32khz on F7 maxes CPU in bf3.2

Cheredanine, 2017-06-01:
Nero F7 with bf3.2 Dev builds
Enabling 32khz maxes CPU load 

Odd because bluejay F4 runs 32/16 at somewhere between 50 and 60%

Perhaps the filtering that gets stripped out on f4 with 32khz enabled is left in for f7?

jflyper, 2017-06-01:
#3050 (https://github.com/betaflight/betaflight/issues/3050#issuecomment-303294438)

Cheredanine, 2017-06-01:
Yeah I see, in now way related to title of that issue, if it is gonna be a catch all f7 issues then it would be helpful for it to be named as such

jflyper, 2017-06-01:
@Cheredanine No offense. Just wanted to be informative ;)

Cheredanine, 2017-06-01:
@jflyper 
Non taken :) ty

kc10kevin, 2017-06-02:
I can confirm the same behavior with the FuryF7.   Something amiss.

MJ666, 2017-06-03:
Also confirm on AlienFlight F7.

MATEKSYS, 2017-06-04:
Confirm on our F722 demo board.

blckmn, 2017-06-04:
Confirmed. I am looking at the SPI speed as the culprit.

aniksuper, 2017-06-06:
confirm on Omnibus f7 board. cant reach even 16/16.  only 16/8 or 10.6/10.6 are okay

bnn1044, 2017-06-07:
Confirm on CLRACINGF7 board. 16/8K is about max 100%

brucesdad13, 2017-06-10:
I modified one of my AlienWhoop F4 and replaced the STM32F405RGT6 with an STM32F722RET6 (bypassed old VCAP to VSS and put 4.7uF cap for pin 30 VCAP_1). On the F4 I am able to run 32kHz gyro sampling and 16kHz PID loop with plenty of CPU to spare. 32/32 on the F4 is too much and goes to 100%. tasks output shows PID loop is using the bulk of the CPU. One would assume the F7 having higher clock speed etc would outperform the F4 but perhaps due to CMSIS differences or some bug performs much worse. I'm not even able to sustain 32/16. Happy to give any other information that might be useful so we can get to the bottom of this F7 issue.

MJ666, 2017-06-10:
The AlienFlight F7 will also run with 32/16. Likely the code need some more optimization.

borisbstyle, 2017-06-10:
There is some waiting in SPI reads from gyro where this performance is
lost. That is the place that needs optimalisation if you dont want to
overclock.
Overclocking is another option.

aniksuper, 2017-06-10:
Will be the optimisations done in summer? or we have to wait release in october-november

Cheredanine, 2017-06-10:
Issue is marked for 3.2, expect the guys will sort it in that release or subsequent point release.

@borisbstyle 
Reported as tested, tbh I don't notice any difference in flight higher than 8/4

jflyper, 2017-06-12:
@brucesdad13 
>I modified one of my AlienWhoop F4 and replaced the STM32F405RGT6 with an STM32F722RET6 (bypassed old VCAP to VSS and put 4.7uF cap for pin 30 VCAP_1). 

Just curious how you managed the 1-pin shift for pins 25-29.

brucesdad13, 2017-06-13:
@jflyper luckily the only pin I was using in that range was 28 for spektrum bind. 😄

jflyper, 2017-06-13:
@brucesdad13 😄

sensei28, 2017-07-01:
waiting for optimisation for f7 board. i dont want to wait all summer pls

Cheredanine, 2017-07-01:
@sensei28 
This is open source, feel free to learn coding and optimise it as you require

sensei28, 2017-07-01:
ok can you help me about which ide  and compiler do you use  and how to compile code? And do you use development board or simulator for testing?

brucesdad13, 2017-07-01:
@sensei28 I think most of the devs are using nucleo boards but you can more cheaply get a flight controller if you're only testing betaflight. The build environment you need an arm cross compiler. https://github.com/cleanflight/cleanflight/tree/master/docs/development

martinbudden, 2017-07-01:
See

PR https://github.com/betaflight/betaflight/pull/3381 - Make SPI read and write register generic
PR https://github.com/betaflight/betaflight/pull/3399 - Cache SPI handle to improve performance for F7 boards
PR https://github.com/betaflight/betaflight/pull/3402 - Improved efficiency of gyro read for SPI

blckmn, 2017-07-01:
@sensei28 depends on what operating system you use.

I use Visual Studio, Visual Studio Code, and Sublime.

mikeller, 2017-07-01:
Eclipse and vim under linux here. But very interested to hear if VS code turns out to be any good for the task.

sensei28, 2017-07-05:
i am using macOS sierra. i think eclipse is the best solution for me.

madchiller, 2017-07-20:
Not sure if it's related or if I should start a new thread.  Today I tried the latest build #1484 e0a7594 and when I select 32khz and hit save and reboot.  The option is unchecked and not saved.... Was it disabled?  Using configurator 3.1.2

Cheredanine, 2017-07-20:
@madchiller 
Which flight controller?

madchiller, 2017-07-20:
Sorry. Omnibus F7.

wind0r, 2017-07-20:
@madchiller this happens if the gyro do not support 32khz
since the board has 2 gyros you need to select the icm one first.. default gyro is the mpu6000 (you can check this with typing status in cli)
you need to use the cli and write 'set gyro_to_use = 1' and 'save'
then reboot and type status in cli and check if now the icm gyro is selected

madchiller, 2017-07-20:
Oh shoot. I knew this... Just early, not enough coffee, I and forgot...
Thank you!

madchiller, 2017-07-20:
OK, Well I can confirm that i am also seeing 100cpu load with 32khz enabled.  Now I'm struggling to turn it off.  May have to reflash.

madchiller, 2017-07-20:
On Flip32 Omnibus F7    16K/16K gives about 68 percent CPU.  

16k/ 8k gives about 25 percent CPU, gonna rock this for now

DieHertz, 2017-08-11:
Tune rates down, no 32KHz doesn't mean there is no F7 support :-)

wind0r, 2017-08-11:
Running 32khz will require removing the HAL. People are working on that but i will not be released with 3.2.

madchiller, 2017-08-11:
@wind0r Thank you for the update.  I as just curious to,the status.  The quad is flying great as is... Best!

RipperDrone, 2017-08-12:
just to understand it better: The (overclocked) F4 can handle the HAL commands faster than the F7? Or is the HAL for F7 not developed to the same speed performance as for f4?

DieHertz, 2017-08-12:
Something is wrong, we need a more thorough research. For now it seems that every operation on F7 has some hidden overhead which makes tasks like sampling gyro over SPI more expensive.

RipperDrone, 2017-08-12:
Ack. Thank you for explaining - makes it easier to accept the extended wait for F7 FC's catching up with F4's in performance :-)

MATEKSYS, 2017-08-13:
F722 & 20689,  acc ON,  OSD ON 
16K/8K, 27%,  
32k/8K  60%
16K/16K, 100%

yoodog, 2017-10-14:
big up to the issue! any update? :( have omnibus f7 v2 from airbot

OBE-Pilot, 2017-10-18:
Also seeing this issue on Omnibus f7 V2 with latest build, any work currently being done to this?

VoicOfReason, 2017-11-03:
I'm having the same issue at 32khz on Omnibus F7 V2 board.

wind0r, 2017-11-03:
As @DieHertz allready wrote 5 post above and I wrote some more post above every F7 has this "Issue"
Feel free to find the overhead and help replacing the HAL to speed stuff up.

martinbudden, 2017-11-03:
I think the next thing that needs to be addressed as part of this is getting the Tightly Coupled Memory (TCM) set up correctly. There is some work going on in this area in iNav which could be reused/ported over.

Brisingrox, 2017-11-12:
Confirm that FortiniF7 also at 100% once 32k activated.

wind0r, 2017-11-12:
As multply post above allready say every F7 board has the "issue".
There totaly is no need to list every flightcontroller which is out.

If you want to help get into the code and get your hands dirty or ask the manufacture to do it.
Or ask the manufacture to pay a developer to do it or help with donation.

Remember Betaflight is a hobby project and everyone here just is doing it in his spare time and most Companys simply use this worked to get your money. Ask those companys first!

RipperDrone, 2017-11-12:
I feel you and understand your comment - everybody is complaining about missing F7 performance but most of them do not contribute any efforts into BetaFlight development themselves. Even more, some companies are capitalising on BF being a great piece of FC code. But please also understand that 99% of us users are NOT capable of rewriting a HAL, because it requires much in-depth hardware knowledge, STM MC insight etc. 
I wish I could contribute something to the F7 32k problem, but it's way beyond my knowledge.

Likewise I wish BF could exploit the 32k capability to its full extent, and seeing this limited to some F4 FCs only (and not yet stable afaik) is a pain. Does it matter to flight performance? I don't know. But lagging behind copymakers like RF in this aspect is somehow aching...

Brisingrox, 2017-11-12:
Same here it wasn't my intention to complain but just to inform. I would really like to help you rewrite some code but it's out of my range of knowledge. Besides this more then sending some Paypal to the dev's as support is  the only way I know to get you guy's some help. I greatly appreciate all the effort and time all the dev's put into this great Project. You guy's are awesome and you have all my support I can give

Brisingrox, 2017-11-12:
By the way. Can someone explain to me why the overclock feature isn't working with the F7?

wind0r, 2017-11-12:
Similar to F3 you can overclock the F7 a little but you do not have a Clockdivider for USB so USB will not work. F4 doesnt have this problem.

F7 optimization is on the milestone list for 3.3 https://github.com/betaflight/betaflight/projects/3

Brisingrox, 2017-11-12:
Thanks for the information. Looking forward to this and will keep a eye on all the next pre release Betaflight version's for testing purposes

jcutcher01, 2018-01-27:
I love BF and currently all of my quads are F4s running 3.2.4. I am hoping 3.3 will get optimized for the F7.

jcutcher01, 2018-02-02:
Ok man BF 3.4 milestone? Ouch

automatt1c, 2018-02-24:
The filter optimizations that are in 3.3 are going to help the f7 more than running 32k anyway. With 3.3 32k is the first time it's actually running good.

DieHertz, 2018-02-24:
Not sure what you mean, could you elaborate?

automatt1c, 2018-02-24:
Before the new kalman and biquad RC+FIR filters were created in 3.3. 32k Just ran bad or at least worse than 8k/8k. Now that we have proper stage 2 filters for 32k they can focus on optimizations. You can use the new stage 2 filters on the f7 at 8k/8k and its going to make it fly even better as well.  thats why it was pushed out to 3.4.

DieHertz, 2018-02-24:
So you mean 32KHz was meaningless before, that's mostly true.
As about F7 performance, me and @jflyper are working on it and have some nice progress. Not finished yet though.

automatt1c, 2018-02-24:
That great news!

OBE-Pilot, 2018-02-24:
That is fantastic news.

[4 more replies]

closed 2018-04-21 by DieHertz