betaflight/betaflight #3203: sp racing evo fault

waynewendrone, 2017-06-04:
i connected my sp racing evo up to my quad and flashed it with beta flight, then the next day i come back the board only has the 5v ok led on (the green one) when powered up and is not connecting or showing up as a com port anymore. I'm getting around 3.7v on uart 2 and ppm and 2.2 v on spectrum satellite port. please advice me what to do as I've tried everything. the Stm 32 cpu also heats up really fast when power runs through the board

Arakon, 2017-06-04:
That's a hardware issue and has nothing to do with betaflight. You likely have a fried voltage regulator on the board.

waynewendrone, 2017-06-04:
Ok what should I do... should I look for a refund??

closed 2017-06-04 by martinbudden