betaflight/betaflight #3211: yaw drift LUXV2_RACE

adrianmiriuta, 2017-06-05:
yaw drift LUXV2_RACE (static on table without arming , no vibration issue)
60 deg /10 minutes
standard gyro settings 

changing 
fc_acc = calculateAccZLowPassFilterRCTimeConstant(11.0f);
reduces drift to 30 deg/10 minutes.

Reducing gyro refresh rates also reduces drift 
with standard setting of  fc_acc.

Any idea how to eliminate

ledvinap, 2017-06-05:
@adrianmiriuta : Your results are in line with gyro specification, don't expect to get significantly better results. Also actual flight results will be much worse.

adrianmiriuta, 2017-06-05:
@ledvinap 
sorry I don't get it.
If changing from 1kHz gyro refresh rate to 8kHz ,increases gyro drift 4x fold, 
it means to me that with higher refresh rate we integrate more gyro noise.
And with changing   calculateAccZLowPassFilterRCTimeConstant we cutoff some of this noise.
Should the LowPass filter time constant not be linked to the refresh rate ?
Or at least be configurable in the CLI ?

ledvinap, 2017-06-05:
From mathematical perspective, integrator itself is low-pass filter, it should remove all higher-frequency components.
The drift you get is from non-gaussian noise, nonlinearities, gyro imperfections and (probably) discretization. Filtering may help a bit, but has little use in quadcopter firmware context.

adrianmiriuta, 2017-06-05:
@ledvinap

in my setup LUXV2_RACE
standard refresh rates

changing this
    //fc_acc = calculateAccZLowPassFilterRCTimeConstant(5.0f); // Set to fix value
    fc_acc = calculateAccZLowPassFilterRCTimeConstant(11.0f);

actually halves yaw drift !

ledvinap, 2017-06-05:
It is a bit strange, because `fc_acc` is RC filter for Z axis measurement, used only for Z velocity/altitude estimation. So is may be only measurement error (but there may be feedback affecting it).
But even if it does affect yaw calculation, it changes drift from zero to zero, so there is no point spending time on it.

Can you try similar test with motors running? Even test without propellers should do, unless your motors are perfectly balanced.

adrianmiriuta, 2017-06-06:
Tere is no big difference with motors running  (+20%).
The readings are not very stable (10-20%)  variation seem to be normal.
 
I think my issue is similar to 
https://github.com/betaflight/betaflight/issues/399

I also got it when i switched to firmware 2.1.7.

closed 2018-04-21 by DieHertz