betaflight/betaflight #3213: 3.1.7 Configuration is reset during savings in the Configurator

Faduf, 2017-06-05:
Hi,

Due to the commit 837f24f098c5216bde9d4e881f3b3639a319a303, the configuration reset when I click on the save buttons in the PID Configuration tab from the Configurator. This is happening on my board "YupiF4" with a STM32F405RGT6.
No more problem when I use the src/main/config/config_eeprom.c file from the 3.1.6 release.

Thanks,

mikeller, 2017-06-05:
Hmm, that is odd, this does not seem to be affecting any of the other STM32F405 targets. Can you try reflashing with 'Full chip erase' in order to make sure it's not a problem with the existing configuration?

Faduf, 2017-06-06:
I tried again, flashing with the "Full chip erase" option and I can't reproduce the problem. It was maybe a conflict between the old and the new flash memory mapping. Thanks!

closed 2017-06-06 by Faduf

AndersHoglund, 2017-06-06:
Slightly OT Q.  Flashing an F4 with "Full Chip Erase", does that work now? If so, when and how was that fixed??  Ref: https://github.com/betaflight/betaflight-configurator/issues/200

mikeller, 2017-06-06:
I couldn't tell you. Using Chromium on linux, flashing with 'Full chip erase' on F4 always worked for me, except for a very few cases where it timed out, so I suspect it's something that is dependent on some timer in Chrome, and probably dependent on the Chrome version as well.